ARM-software/bsa-acs #3: output is nightmare

hrw, 2021-05-23:
Current output of BSA ACS in unreadable mess. Default run of tool should output just ID/name of test and it's result:

```
      ***  Starting GIC tests ***                                                                                 
Operating System:                                                                                                 
 201 : B_GIC_01: Check GIC version                 : Result:  PASS
 202 : B_GIC_02: Check GICv2 Valid Configuration   : Result:  SKIPPED
 203 : B_GIC_03: If PCIe, GICv3 then ITS, LPI      : Result:  FAIL
 204 : B_GIC_04: Check GICv3 Security States       : Result:  PASS
 205 : B_GIC_05: Non-secure SGIs are implemented   : Result:  PASS
 206 : B_PPI_01: Check PPI Assignments for OS      : Result:  PASS
```

Then second run could be done with increased verbosity level to see warning, debugs etc. information.

For SBSA-ACS I did some serious changes in how output is generated: https://github.com/ARM-software/sbsa-acs/pull/146 - would be nice to see something like that in BSA-ACS as well.

hrw, 2021-05-27:
Have you thought about merging SBSA-ACS with BSA-ACS?

They share requirements (as can be seen in [my BSA-SBSA checklist table](https://marcin.juszkiewicz.com.pl/download/tables/bsa-sbsa.html)) and you would have one project to maintain.

prasanth-pulla, 2021-05-27:
This BSA repo is only at Alpha quality. The SBSA repo will be merged/leverage code from this repo after we reach a v1.0 REL quality here.

chetan-rathore, 2021-09-30:
Hi @hrw, 

We have made some improvements in the Log output with the recent 1.0 Release. 
The default run of acs will output ID, DESC, and RESULT.
To aid in debugging, only for FAILED test extra debugging message will be printed.

hrw, 2021-09-30:
Looks like I need to check how it looks now ;D

hrw, 2021-09-30:
Heh. Again one step ahead, three steps back...

BSA ACS 0.9:
```
      ***  Starting PE tests ***  
Operating System:
   1 : B_PE_01:  Check Arch symmetry across PE    : Result:  PASS 
   2 : B_PE_02:  Check for number of PE           : Result:  PASS 
   3 : B_PE_03:  Check for AdvSIMD and FP support : Result:  PASS 
   4 : B_PE_04:  Check PE 4KB Granule Support     : Result:  PASS 
   5 : B_PE_05:  Check HW Coherence support       : Result:  PASS 
   6 : B_PE_06:  Check Cryptographic extensions   : Result:  PASS 
   7 : B_PE_07:  Check Little Endian support      : Result:  PASS 
   8 : B_PE_08:  Check EL1 and EL0 implementation : Result:  PASS 
   9 : B_PE_09:  Check for PMU and PMU counters   : Result:  PASS 
  10 : B_PE_10:  Check PMU Overflow signal        : Result:  PASS 
  11 : B_PE_11:  Check num of Breakpoints & type  : Result:  PASS 
  12 : B_PE_12:  Check Synchronous Watchpoints    : Result:  PASS 
  13 : B_PE_13:  Check CRC32 instruction support  : Result:  PASS 
  14 : B_PE_15:  Check PAuth if implementation    : Result:  PASS 
```

BSA ACS 1.0:
```
      ***  Starting PE tests ***
Operating System View:
   1 : Check Arch symmetry across PE              : Result:  PASS
   2 : Check for number of PE                     : Result:  PASS
   3 : Check for AdvSIMD and FP support           : Result:  PASS
   4 : Check PE 4KB Granule Support               : Result:  PASS
   5 : Check HW Coherence support                 : Result:  PASS
   6 : Check Cryptographic extensions             : Result:  PASS
   7 : Check Little Endian support                : Result:  PASS
   8 : Check EL1 and EL0 implementation           : Result:  PASS
   9 : Check for PMU and PMU counters             : Result:  PASS
  10 : Check PMU Overflow signal                  : Result:  PASS
  11 : Check num of Breakpoints and type          : Result:  PASS
  12 : Check Synchronous Watchpoints              : Result:  PASS
  13 : Check CRC32 instruction support            : Result:  PASS
  14 : Check PAuth if implementation              : Result:  PASS
```

Restore IDs from spec. They were useful.

chetan-rathore, 2021-10-04:
Hi @hrw,

The Rule ID was omitted from the test description since we have tests that cover multiple rule ids (ex in power_wakeup, watchdog,  smmu, pcie, exerciser modules), which causes overall output to be unaligned.  

We do print Rule ID for failed tests with debug verbosity.

Thanks!