ARM-software/bsa-acs #11: PENDING result from Test 501 : Wake from Watchdog WS0 Int

sunnywang-arm, 2021-12-14:
We run into the failure below on some ARM systems.
`501 : Wake from Watchdog WS0 Int            : Result:  A01F5000 `

@semihalf-bernacki-grzegorz found the root cause. The following is my understanding of the root cause, feel free to correct me, Greg.

The problem is that the 501 test checks the result too early, so it gets PENDING result (A01F5000). It directly checks the result right after setting the WS0 timer (without waiting for the WS0 timer to get expire). In other words, the problem is that the WS0 timer interrupt service routine hasn’t got a chance to be triggered to update the result.   

@semihalf-bernacki-grzegorz is working on a fix to BSA by adding code to wait for the WS0 timer to get expire. 

@chetan-rathore  if you have any concern about the fix, feel free to let us know.

semihalf-bernacki-grzegorz, 2021-12-14:
Yes, you're correct, the problem is that any other interrupt (secure) can wake up a core and result is being checked to early. I will modify my patch according to your suggestions and create a PR.

chetan-rathore, 2021-12-15:
Hi @sunnywang-arm/ @semihalf-bernacki-grzegorz, 

In the wake-up tests, PE is put in WFI mode, and then the test waits for the respective interrupt to get generated.
In 501, test is waiting for Watchdog interrupt. 

Can you please give details on which interrupt(secure) was generated before the WD interrupt and why. That will help us to review/come up with any fix if required.

Thanks!

semihalf-bernacki-grzegorz, 2021-12-20:
Hi @chetan-rathore, the problem is timer interrupt which is used to poll link status and handle network interfaces. Following patch solves the problem (but maybe delay in the loop can be shortened):
[wd_test.patch.txt](https://github.com/ARM-software/bsa-acs/files/7745924/wd_test.patch.txt)

chetan-rathore, 2021-12-23:
Thanks @semihalf-bernacki-grzegorz , 

We are reviewing the patch and will update soon.

chetan-rathore, 2021-12-24:
Hi @semihalf-bernacki-grzegorz,

The current patch misses the intent of the test. The PE entry to WFI mode is removed in the patch but the test needs PE to enter in idle mode and then PE to resume execution after receiving WD interrupt (BSA 1.0 Spec: Table 8  Semantic B).

The present issue is due to timing issue between WD interrupt timeout and the timer interrupt (which is used to poll link status and handle network interface) timeout. 

To further debug we want to check system timer counterfrequency and also status of other power and wakeup test. 
Can you please share BSA log with -v 1 verbosity.   (PCIe test can be skipped)

> Bsa.efi -skip 800 -v 1

Thanks,
ACS team

semihalf-bernacki-grzegorz, 2022-01-04:
Hi @chetan-rathore, I apologize for late response. You can just use val_power_enter_semantic(BSA_POWER_SEM_B) instead of val_time_delay_ms() inside the while() loop. I also tried this version and it worked for me. 
Here is the log from run with '-v 1' (I just removed some PCIe debugs to make it shorter):
[bsa_verbose.log](https://github.com/ARM-software/bsa-acs/files/7806291/bsa_verbose.log)

chetan-rathore, 2022-01-07:
Hi @semihalf-bernacki-grzegorz,

Thanks for the logs and patch.
As per BSA specification B_WAK_10 (Semantic B), PE can wakeup on receipt of any interrupt. 
Test is updated to consider the case where the PE has waked up from any interrupt (other than WD or failsafe) as success.

Changes for 501 test + SMMU hang issue #12 -> https://github.com/chetan-rathore/bsa-acs/commit/bd24ef61c168aea663958b96ec852992bb5a4757
 
Thanks,
ACS team

chetan-rathore, 2022-01-12:
Hi @semihalf-bernacki-grzegorz @sunnywang-arm ,

The changes are updated as part of #15

sunnywang-arm, 2022-01-12:
Closing it as the fix got merged and verified. Thanks, @semihalf-bernacki-grzegor and @chetan-rathore.

closed 2022-01-12 by sunnywang-arm