ARM-software/bsa-acs #14: 802 : PE - ECAM Region accessibility check - Failure - First device in a ECAM space is not a valid device

sunnywang-arm, 2022-01-11:
There is an assumption in the test suite that the first device will always be at Bus 0 Dev 0 Func 0. However, it is possible to see that the first device in ECAM space is an RP (Bridge device) having a non-zero BDF number, and there is no requirement in the PCIe spec to support this assumption. Therefore, we need to update the code to remove the assumption. 

We also confirmed that VMware and most Linux distros are working fine with the case that the first device has a non-zero device number.

chetan-rathore, 2022-01-30:
Hi @sunnywang-arm,

The Bus 0 Dev 0 Func 0 check has been removed from test #16 and updated further with #18  changes

sunnywang-arm, 2022-01-31:
Yeah, confirmed. #16 fixed the issue. Thanks, @chetan-rathore and @Sujana-M.
Closing this issue.

closed 2022-01-31 by sunnywang-arm