ARM-software/bsa-acs #21: test_os_p030.c

ndhillonm, 2022-02-06:
In the 'esr' function of this test, ELR register is being set to an instruction to jump to. It is not working for me and I am wondering if this is a bug.
Current instruction is as follows but results in a hang in our system.
"val_pe_update_elr(context, (uint64_t)branch_to_test);"

To get around this, "val_pe_update_elr(context, (uint64_t)(((EFI_SYSTEM_CONTEXT_AARCH64*)context)->ELR + 4));", seems to behave better.

Is this a real issue or is it just our SoC ?

Thanks,

ndhillonm, 2022-02-27:
> External Email
>
> Hi @ndhillonm<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_ndhillonm&d=DwMCaQ&c=nKjWec2b6R0mOyPaz7xtfQ&r=1o2VA7xKsHvmQ4HFRY2JZggF-gGh1t4bxHFABsB1WTo&m=qnkNXnESac38p9pJw3epqv4z8rcQK5CWd_frG7JiOC0KONWFvtA4kg-3SZW6sHw3&s=ra65t6zZQDhp40UowR75fr2_ONkG32YHcX0wy-sYysg&e=>,
>
> The "branch_to_test" is set to jump to the desired location in the test after handling the exception. val_pe_update_elr function updates the ELR register to the offset pointed by branch_to_test.
>
> Can you please share debug logs with this binary to debug the issue further: https://github.com/chetan-rathore/bsa-acs/blob/main/prebuilt_images/Bsa_430.efi<https://urldefense.proofpoint.com/v2/url?u=https-3A__github.com_chetan-2Drathore_bsa-2Dacs_blob_main_prebuilt-5Fimages_Bsa-5F430.efi&d=DwMCaQ&c=nKjWec2b6R0mOyPaz7xtfQ&r=1o2VA7xKsHvmQ4HFRY2JZggF-gGh1t4bxHFABsB1WTo&m=qnkNXnESac38p9pJw3epqv4z8rcQK5CWd_frG7JiOC0KONWFvtA4kg-3SZW6sHw3&s=aE2DVxas_NPdoAtVl-y1gT8QRQ0yIUNJRTtOOQMhJgc&e=>
>
> We have added prints to check branch_to_test offset value in payload and esr function.
>
> Thanks!!

Hi Chetan,

Log attached. Hope you receive it.

Thanks,
Narinder

chetan-rathore, 2022-03-02:
Hi @ndhillonm, 

Unfortunately, didn't receive the logs on the github account email id.
Can you please send the logs to chetan.rathore@arm.com

Thanks!!

ndhillonm, 2022-03-02:
Sent.

chetan-rathore, 2022-03-03:
Hi @ndhillonm,

Thanks for the logs.

From the test 830 prints, val_pe_update_elr is working correctly.
All the PCIe tests have also been completed.

On the failure of test 830 in the logs, it seems the root port URD (unsupported request detected) bit is not getting set.

As per PCIe spec,
Memory Space Enable - Controls a Function's response to Memory Space accesses. **When this bit is
Clear, all received Memory Space accesses are caused to be handled as Unsupported Requests.** When
this bit is Set, the Function is enabled to decode the address and further process Memory Space

chetan-rathore, 2022-03-27:
Hi @ndhillonm, 

Please reopen the ticket, if the issue is observed again.

closed 2022-03-27 by chetan-rathore