ARM-software/bsa-acs #24: System hangs at test 861 : PCIe Unaligned access

sunnywang-arm, 2022-03-22:
In the verbose messages (the outputbsa -v 1), test 861 keeps printing the same message for checking bdf 000000. 

```
[ 601.380221] 861 : PCIe Unaligned access START
[ 601.386824]
[ 601.386824] Calculated config address is 28c0600010
[ 601.386824] The BAR value of bdf 060000 is 3011
[ 601.386824] Calculated config address is 28c0000010
[ 601.386824] The BAR value of bdf 000000 is 0
[ 601.386824] Calculated config address is 28c0000010
[ 601.386824] The BAR value of bdf 000000 is 0
[ 601.386824] Calculated config address is 28c0000010
[ 601.386824] The BAR value of bdf 000000 is 0
[ 601.386824] Calculated config address is 28c0000010
```

It looks like the problem is that there is no break in while loop in https://github.com/ARM-software/bsa-acs/blob/1cc33fea036e4a34dae7f75366e685226a647417/test_pool/pcie/operating_system/test_os_p061.c#L50

The same test (405) in sbsa doesn't have this issue because it has a condition check to break the loop in https://github.com/ARM-software/sbsa-acs/blob/5ccf09073bd4c17cab96bc338e2a3a314bf3a078/test_pool/pcie/test_p005.c#L84

Therefore, we may need to make the change below to fix this issue. 
1. Move the line "next_bdf:" to above the line "while (count--) {"
2. Remove "count--;" for the places that would run "goto next_bdf;"

gowthamsiddarthd, 2022-04-14:
This issue has been solved with the PR: https://github.com/ARM-software/bsa-acs/pull/27

chetan-rathore, 2022-04-18:
Hi @sunnywang-arm,

The fix is merged with #27.

Thanks,
ACS team

closed 2022-04-18 by chetan-rathore