ARM-software/bsa-acs #25: BSA test 605: Memory Attribute of DMA failure

sunnywang-arm, 2022-03-24:
Note that this may be related to the Linux kernel as we don’t see this failure in ACS 3.1. ACS 3.1 doesn’t detect any DMA controller.   
 
We got the verbose message below with this failure, which doesn't give us enough information to fix this issue and it may probably be a test suite issue or expected behaviour (as testing this may require a device-specific driver). Could anyone give us more information?  

```
In decode_mem_attr_sh with attr=ff, sh=0
    DMA controler 1: IO Coherent DMA memory should be inner/outer writeback, inner shareable
In decode_mem_attr_sh with attr=ff, sh=0
    DMA controler 0: IO Coherent DMA memory should be inner/outer writeback, inner shareable
```

- The message shows that pal_dma_mem_get_attrs() gets invalid Memory attributes and Sharability info, so the DMA entry in g_dma_info_table may be invalid. How does sbsa/bsa build the g_dma_info_table? Is the information in the pal_dma_create_info_table from MMU translation table or from APCI table? Where is the pal_dma_create_info_table source code?
- Is the failure a firmware issue or Hardware issue? 
- Is there a Linux command to check DMA controller?  It looks like all the devices are working fine under OS.

chetan-rathore, 2022-06-22:
Hi @sunnywang-arm, 

We are working on a similar test in SBSA and verifying the fix for both BSA and SBSA. We will share the updates soon.

Thanks,
ACS

chetan-rathore, 2022-06-28:
Hi @sunnywang-arm, 

Changes are raised on gitlab: https://gitlab.arm.com/linux-arm/linux-acs/-/merge_requests/18

Thanks,
ACS

chetan-rathore, 2022-06-29:
Hi @sunnywang-arm, 

The changes have been merged.

Thanks,
ACS

closed 2022-06-29 by chetan-rathore

sunnywang-arm, 2022-10-11:
The fix is in https://gitlab.arm.com/linux-arm/linux-acs/-/commit/379cd445ddf264a2b1506dd1ba841e1fb81eb2f2