ARM-software/bsa-acs #30: BSA Linux 104 : Addressability

sunnywang-arm, 2022-05-19:
This is the same issue as the one with sbsa-acs in https://github.com/ARM-software/sbsa-acs/issues/234

The error message is the following
`WARNING:The device with bdf=0xXXXXXXX doesn't support 64 bit addressing and is not behind SMMU. Please install driver for this device and test again.`

The root cause is that the ACS Linux kernel doesn't have the device driver for the device with bdf=0xXXXXXXX.

Suggest skipping this test in automation. People who have a need to run this test can still try to find and install the device driver and then manually run it.

If we still need to keep this test in automation, the suggestion would be to add device drivers as many as possible. Otherwise, it is easy to run into this failure when the system has an integrated PCIe device or a PCIe card.

chetan-rathore, 2022-06-22:
Hi @sunnywang-arm, 

We are tracking this as part of SBSA #234.

Thanks,
ACS

chetan-rathore, 2022-07-07:
Hi @sunnywang-arm,

The changes are merged.

Thanks,
ACS

closed 2022-07-07 by chetan-rathore

sunnywang-arm, 2022-08-09:
Fixed by https://git.gitlab.arm.com/linux-arm/linux-acs/-/commit/379cd445ddf264a2b1506dd1ba841e1fb81eb2f2