ARM-software/bsa-acs #40: Test cases like 816 for RCiEP and i-EP should not be run

sunnywang-arm, 2022-07-26:
This is the same issue as https://github.com/ARM-software/sbsa-acs/issues/256

The requirements for RCiEP and i-EP (based on B_REP_1 and B_IEP_1 in BSA spec) are referenced in SBSA Level 6 checklist in SBSA spec rather than the BSA checklist. Therefore, we should not test these requirements in BSA.
For example, we should skip test case "816: Check MSI and MSI-X support rule" that is based on B_REP_1 and RE_INT_1.

sunnywang-arm, 2022-08-09:
Note that the test cases below need to be updated as well. 
812 : Check Power Management rules 
828 : Power management cap rules
829 : Power management/status rule

chetan-rathore, 2022-09-05:
Hi @sunnywang-arm,

Changes are raised for the issue.

Thanks,
ACS team

closed 2022-09-05 by chetan-rathore