ARM-software/bsa-acs #42: Test 601 improperly failing USB 1.1 controllers

PhatFree, 2022-08-10:
According to RB_PER_01 & RB_PER_02 of BSA, If the system has a USB2.0/USB3.0 host controller, it needs to be EHCI/XHCI  complaint.

But test 601 is just testing all USB controllers to see if they're XHCI or EHCI complaint, improperly failing USB 1.1 controllers that are not complaint with either.

https://github.com/ARM-software/bsa-acs/blob/79986b026a923550b74783e8376b7126bb2c63d4/test_pool/peripherals/operating_system/test_os_d001.c#L47

samerhaj, 2022-08-15:
In other words, the  BSA Spec does not forbid UHCI (USB 1.1) host controllers form existing in a compliant system. All what the spec is saying is that If USB 2.0 or USB 3.0 host controllers are present, they must be compliant to the EHCI or XHCI specifications. 

The test is not really checking the compliance of these rules. As it is written, every failure to this test is going to be a false positive, an the test cannot really detect non-compliance of these BSA rules.

chetan-rathore, 2022-08-25:
Thanks @PhatFree and @samerhaj for your inputs.

We are checking the issue on our side.

Thanks,
ACS team

chetan-rathore, 2022-09-24:
Hi @PhatFree,

The changes are merged.

Thanks,
ACS team

closed 2022-09-24 by chetan-rathore