ARM-software/bsa-acs #43: Test case 606 assuming system clock reference for UART

PhatFree, 2022-08-11:
https://github.com/ARM-software/bsa-acs/blob/79986b026a923550b74783e8376b7126bb2c63d4/test_pool/peripherals/operating_system/test_os_d005.c#L131

In test 606, we perform two checks for the baud rate. The first by grabbing the info from SPCR. The second by trying to compute it via UART registers and the system clock. The issue with this is that the UART doesn’t have to use the system clock as its reference, so if a system is using a different clock reference it will fail this test, even if the system is indeed complaint.

I suggest this second check for the baud rate is removed

robbiek-xsl, 2022-08-22:
Also recently encountered this issue and did some research.  The latest SPCR specification (revision 3) from Microsoft supports providing the clock frequency:

https://docs.microsoft.com/en-us/windows-hardware/drivers/serports/serial-port-console-redirection-table

However I do not believe support has been upstreamed to the Linux kernel.  I assume since Microsoft added it that Windows does support it.  As best that I can tell looking through Linux code the frequency is assumed to be 1.8432 MHz but I have not had a chance to verify this on HW.

chetan-rathore, 2022-08-25:
Thanks @PhatFree and @robbiek-xsl for your inputs.

We are checking the issue at our side.

Thanks,
ACS team

gowthamsiddarthd, 2022-09-05:
Hi @PhatFree/ @robbiek-xsl,

Thank you for reporting this.
This has been fixed as part of https://github.com/ARM-software/bsa-acs/pull/52

Regards,
Gowtham Siddarth
ACS Team

closed 2022-09-06 by chetan-rathore