ARM-software/bsa-acs #47: Test case 830 hangs in an endless loop with any GCC optimization enabled

robbiek-xsl, 2022-08-23:
The BSA (and SBSA equivalent) both hang in an endless loop unless compiler optimization is completely disabled.  These tests use a very questionable programming technique of taking the address of a C label and having the exception handler return to that address.  With optimization enabled GCC is free to do whatever it wants with the code flow including possibly duplicating code paths, etc.  

The general GCC issue is discussed in this thread for instance: 

https://stackoverflow.com/questions/6127563/label-moving-after-optimization-is-turned-on-in-gcc 

At a minimum I would suggest specifically disabling optimization in bsa-acs/uefi_app/BsaAcs.inf.

```
       Received exception of type: 0
       Calculated config address is 69400004 pal_mmio_read Address = 69400004  Data = 100004 
 pal_mmio_write Address = 69400004  Data = 100004 

       Calculated config address is 69400034 pal_mmio_read Address = 69400034  Data = 48 

       Calculated config address is 69400048 pal_mmio_read Address = 69400048  Data = 424010 

       Calculated config address is 69400050 pal_mmio_read Address = 69400050  Data = 2000 
 pal_mmio_write Address = 69400050  Data = 2000 

       Calculated config address is 69400034 pal_mmio_read Address = 69400034  Data = 48 

       Calculated config address is 69400048 pal_mmio_read Address = 69400048  Data = 424010 

       Calculated config address is 69400050 pal_mmio_read Address = 69400050  Data = 2000 
 pal_mmio_write Address = 69400050  Data = 82000 

       Calculated config address is 69400004 pal_mmio_read Address = 69400004  Data = 100004 
 pal_mmio_write Address = 69400004  Data = 100004 

       Received exception of type: 0
       Calculated config address is 69400004 pal_mmio_read Address = 69400004  Data = 100004 
 pal_mmio_write Address = 69400004  Data = 100004 

       Calculated config address is 69400034 pal_mmio_read Address = 69400034  Data = 48 

       Calculated config address is 69400048 pal_mmio_read Address = 69400048  Data = 424010 

       Calculated config address is 69400050 pal_mmio_read Address = 69400050  Data = 2000 
 pal_mmio_write Address = 69400050  Data = 2000 

       Calculated config address is 69400034 pal_mmio_read Address = 69400034  Data = 48 

       Calculated config address is 69400048 pal_mmio_read Address = 69400048  Data = 424010 

       Calculated config address is 69400050 pal_mmio_read Address = 69400050  Data = 2000 
 pal_mmio_write Address = 69400050  Data = 82000 

       Calculated config address is 69400004 pal_mmio_read Address = 69400004  Data = 100004 
 pal_mmio_write Address = 69400004  Data = 100004 ```

chetan-rathore, 2022-08-25:
Thanks @robbiek-xsl for your input.

We are checking the issue on our side.

Thanks,
ACS team

chetan-rathore, 2022-08-30:
Hi @robbiek-xsl, 

Can you let us know if the RELEASE build was compiled at your end? 
At our end by default, DEBUG build gets complied in which the compiler optimization is disabled.

We will explicitly make changes to disable the optimization.

Thanks,
ACS Team

robbiek-xsl, 2022-08-30:
Hi,

Good point, so far we've been running with DEBUG only.  Thanks for your help!

chetan-rathore, 2022-09-04:
Hi, @robbiek-xsl, 

We have enforced disabling compiler optimization in the BSA and SBSA inf.

Thanks,
ACS team

closed 2022-09-04 by chetan-rathore