ARM-software/bsa-acs #56: Increase a value of PCIE_ECAP_ID_END macro.

semihalf-bernacki-grzegorz, 2022-09-28:
PCIE_ECAP_ID_END is defined as  0x29

https://github.com/ARM-software/bsa-acs/blob/726b990213b17fe466b9c53d751c0f5c11c16c60/test_pool/pcie/operating_system/test_os_p009.c#L28

However according to "[PCI Code and ID Assignment Specification Revision 1.11"](https://pcisig.com/sites/default/files/files/PCI_Code-ID_r_1_11__v24_Jan_2019.pdf) values up to 0x2C are allowed.

chetan-rathore, 2022-09-30:
Hi @semihalf-bernacki-grzegorz,

Thanks for raising the issue, we will check at our side.

Regards,
ACS team

gowthamsiddarthd, 2022-10-06:
Hi @semihalf-bernacki-grzegorz,

This issue has been fixed as part of the PR: https://github.com/ARM-software/bsa-acs/pull/60

Regards,
Gowtham,
ACS Team

closed 2022-10-10 by gowthamsiddarthd