ARM-software/bsa-acs #67: PLATFORM_OVERRIDE_TIMEOUT does not work

bearzly, 2022-11-02:
Hi,

I am trying to use the PLATFORM_OVERRIDE_TIMEOUT to change the timeouts for my platform. However, I noticed that changing this value doesn't actually take effect.

The following diff shows changing the macro value to 1 and then adding a compile error to the #if in pal_interface.h. This should cause the build the fail, but it will still happily compile for the uefi_acpi platform with the "default' timeout values

I would attempt a fix but I'm not sure how the platform overrides are meant to work here

```
diff --git a/platform/pal_uefi_acpi/include/platform_override.h b/platform/pal_uefi_acpi/include/platform_override.h
index ec00b2c..5c4ec8a 100644
--- a/platform/pal_uefi_acpi/include/platform_override.h
+++ b/platform/pal_uefi_acpi/include/platform_override.h
@@ -36,7 +36,7 @@
 #define PLATFORM_OVERRIDE_PLATFORM_TIMER_GSIV 58

 /* Change OVERRIDE to 1 and define the Timeout values to be used */
-#define PLATFORM_OVERRIDE_TIMEOUT        0
+#define PLATFORM_OVERRIDE_TIMEOUT        1
 #define PLATFORM_OVERRIDE_TIMEOUT_LARGE  0x10000
 #define PLATFORM_OVERRIDE_TIMEOUT_MEDIUM 0x1000
 #define PLATFORM_OVERRIDE_TIMEOUT_SMALL  0x10
diff --git a/val/include/pal_interface.h b/val/include/pal_interface.h
index 623c70c..fff6e49 100644
--- a/val/include/pal_interface.h
+++ b/val/include/pal_interface.h
@@ -61,6 +61,7 @@
   typedef UINT64 addr_t;

 #if PLATFORM_OVERRIDE_TIMEOUT
+#error fail
     #define TIMEOUT_LARGE    PLATFORM_OVERRIDE_TIMEOUT_LARGE
     #define TIMEOUT_MEDIUM   PLATFORM_OVERRIDE_TIMEOUT_MEDIUM
     #define TIMEOUT_SMALL    PLATFORM_OVERRIDE_TIMEOUT_SMALL
```

gowthamsiddarthd, 2022-11-04:
Hi @bearzly,

Thank you for reporting this issue.
We are currently working on providing an optimal fix. I will update you once the change has been pushed.

Regards,
Gowtham,
The ACS Team

gowthamsiddarthd, 2022-11-07:
Hi @bearzly,

Fix for this issues has been merged by the PR: https://github.com/ARM-software/bsa-acs/pull/68

Regards,
Gowtham

closed 2022-11-08 by gowthamsiddarthd