ARM-software/bsa-acs #72: Test 828 - power management capability is not requried for BSA complaince

sunnywang-arm, 2022-11-30:
In BSA spec, PCI_IN_05 doesn’t mention that we need to check 7.5.2 (PCI Power Management Capability Structure). 
Therefore, we need to remove test 828.

samerhaj, 2022-12-02:
BSA 1.x has rules for PCI Power Management Capabilities for RCiEP and RCEC on-chip peripheral devices (that are presented as PCIe devices). See for instance: RRE_PWR_1, RRE_PWR_2, RRE_PWR_3, RRE_REG_2, as well as the rules in section F.3.6 and F.3.7.

sunnywang-arm, 2022-12-02:
@samerhaj The rules you mentioned are currently referenced in SBSA Level 6 checklist in SBSA spec rather than the BSA checklist, so we moved the tests to SBSA. The change was tracked by https://github.com/ARM-software/bsa-acs/issues/40. 
Now the BSA test 828 is checking Power management cap of the non-RCiEP device (like non-integrated endpoint devices) which is not required in BSA, so we need to remove the test case.

samerhaj, 2022-12-02:
Thanks for the clarification. Makes sense. If the rules are in SBSA L6, and not mentioned in BSA 1.0 checklist, then they should not be tested as part of the basic BSA ACS, but rather the SBSA ACS.

chetan-rathore, 2022-12-07:
Hi @sunnywang-arm,

Changes are merged for the issue. #75 

Thanks,
ACS team

closed 2022-12-12 by chetan-rathore