ARM-software/bsa-acs #73: Test case 837 - Is accessing the register through PCIIO protocl needed? If so, a improvement might be needed.

sunnywang-arm, 2022-11-30:
Found a question and a improvement while debugging the failure below: 
```
 837 : Check Config Txn for RP in HB                  START
    BDF 0xXXXXXXXXXX
       Reading Class code using PciIo protocol failed 
       Failed on PE -    0
       PCI_IN_12
       Checkpoint --  1                           : Result:  FAIL 
```

- https://github.com/ARM-software/bsa-acs/blob/main/test_pool/pcie/operating_system/test_os_p037.c says the 837 test is for requirement PCI_IN_12, but it looks like we just need to write and read the registers through the ECAM. Is there anything related to PCIIO protocol that I overlooked in PCI_IN_12? If not, should we update the test 837 to match the BSA spec? In other words, I’m not sure if firmware hiding a device is a BSA compliance issue.   

- if we need to catch the mismatch between ECAM and PCIIO access, I’m thinking about a BSA improvement. When the test runs into a mismatch between ECAM and PCIIO, can BSA continue checking other devices instead of stopping the test? In this case, we will not need to modify and build BSA as you suggested to check other devices.

samerhaj, 2022-12-02:
It seems to me that it is fair for ACS to be comparing PCIe reads/writes from different interfaces (ECAM, PCIO, etc..) that are supported by the UEFI/PCIe standards, and confirm that the data is consistent.

The suggested improvement of BSA reporting the failure and continuing to test other devices make sense.

gowthamsiddarthd, 2022-12-13:
Changes to reporting the failure and testing other BDFs https://github.com/ARM-software/bsa-acs/pull/79

closed 2022-12-15 by gowthamsiddarthd