ARM-software/bsa-acs #74: Forever loop when parsing pmu node

thanhhung06, 2022-12-05:
Hi,

I just found that we get the forever loop when the pmu-node of device-tree following the requirement of source code.
Because the `'i'` variable is used in loop, then the condition in `for` is wrong
I changed as below and success to parse, but I have not permission to contribute on this repository.

<pre>--- a/platform/pal_uefi_dt/src/pal_pe.c
+++ b/platform/pal_uefi_dt/src/pal_pe.c
@@ -359,7 +359,7 @@ pal_pe_data_cache_ops_by_va(UINT64 addr, UINT32 type)
 VOID
 pal_pe_info_table_pmu_gsiv_dt(PE_INFO_TABLE *PeTable)
 {
-  int i, offset, prop_len;
+  int i, offset, prop_len, dt_idx;
   UINT64 dt_ptr = 0;
   UINT32 *Pintr;
   int index = 0;
@@ -379,12 +379,12 @@ pal_pe_info_table_pmu_gsiv_dt(PE_INFO_TABLE *PeTable)
 
   Ptr = PeTable->pe_info;
 
-  for (i = 0; i < (sizeof(pmu_dt_arr)/PMU_COMPATIBLE_STR_LEN); i++) {
+  for (dt_idx = 0; dt_idx < (sizeof(pmu_dt_arr)/PMU_COMPATIBLE_STR_LEN); dt_idx++) {
 
       /* Search for pmu nodes*/
-      offset = fdt_node_offset_by_compatible((const void *)dt_ptr, -1, pmu_dt_arr[i]);
+      offset = fdt_node_offset_by_compatible((const void *)dt_ptr, -1, pmu_dt_arr[dt_idx]);
       if (offset < 0) {
-          bsa_print(ACS_PRINT_DEBUG, L"  PMU compatible value not found for index:%d\n", i);
+          bsa_print(ACS_PRINT_DEBUG, L"  PMU compatible value not found for index:%d\n", dt_idx);
           continue; /* Search for next compatible item*/
       }
 
@@ -463,7 +463,7 @@ pal_pe_info_table_pmu_gsiv_dt(PE_INFO_TABLE *PeTable)
               }
           }
           offset =
-              fdt_node_offset_by_compatible((const void *)dt_ptr, offset, pmu_dt_arr[i]);
+              fdt_node_offset_by_compatible((const void *)dt_ptr, offset, pmu_dt_arr[dt_idx]);
       }
   }
 }</pre>

Best Regards,
Hung Huynh

chetan-rathore, 2022-12-05:
Hi @thanhhung06,

Thanks for correcting the loop condition. 
To contribute directly, please raise a PR (pull request) with the changes. You will need to make the changes in your fork and then create a PR to main branch.

Thanks,
ACS team

thanhhung06, 2022-12-07:
Hi @chetan-rathore 

Sorry, I mis-understand the pull request flow on ARM repository and do not notice the "fork".

I created the pull request https://github.com/ARM-software/bsa-acs/pull/76

Best Regards,
Hung Huynh

chetan-rathore, 2022-12-12:
Thanks @thanhhung06 for PR. 

Code has been merged.

closed 2022-12-12 by chetan-rathore