ARM-software/bsa-acs #77: Unable to run explicit test case

shashi-j, 2022-12-08:
On trying to explicitly run test case 839 for non integrated pcie devices,Bsa.efi doesnt run the desired test case but outputs the platform info and skips all tests.Logs are as below

[BSA_TestCase_Log.txt](https://github.com/ARM-software/bsa-acs/files/10187365/BSA_TestCase_Log.txt)
[BSA_Test_Log.txt](https://github.com/ARM-software/bsa-acs/files/10187366/BSA_Test_Log.txt)
[BSA_acpiview_Log.txt](https://github.com/ARM-software/bsa-acs/files/10187367/BSA_acpiview_Log.txt)

chetan-rathore, 2022-12-12:
Hi @shashi-j , 

Thanks for the logs.

Based on the acpiview dump, the system on which you are running is QEMU(sbsa). 

On the QEMU (7.1.0) sbsa-ref at our end, we see it has ony 2 RCiEP devices and 1 PCI device. 
Attached for your ref.  [pci_cmd_log.txt](https://github.com/ARM-software/bsa-acs/files/10210314/pci_cmd_log.txt)
For integrated pcie devices, sbsa acs (https://github.com/ARM-software/sbsa-acs) should be used.

Also the model at our end has no ITS.

To debug the exception during ITS initialization, can you please share steps to obtain the model and the stack for your setup?

shashi-j, 2022-12-12:
Hi Chetan,

I am not testing integrated pcie,hence using bsa-acs.
I applied the test patch below for enabling ITS on top of QEMU(7.1.0) for sbsa-ref platform.

```
diff --git a/hw/arm/sbsa-ref.c b/hw/arm/sbsa-ref.c
index 4bb444684f..c7d19b1a38 100644
--- a/hw/arm/sbsa-ref.c
+++ b/hw/arm/sbsa-ref.c
@@ -63,6 +63,7 @@ enum {
     SBSA_CPUPERIPHS,
     SBSA_GIC_DIST,
     SBSA_GIC_REDIST,
+    SBSA_GIC_ITS,
     SBSA_SECURE_EC,
     SBSA_GWDT_WS0,
     SBSA_GWDT_REFRESH,
@@ -106,6 +107,7 @@ static const MemMapEntry sbsa_ref_memmap[] = {
     [SBSA_CPUPERIPHS] =         { 0x40000000, 0x00040000 },
     [SBSA_GIC_DIST] =           { 0x40060000, 0x00010000 },
     [SBSA_GIC_REDIST] =         { 0x40080000, 0x04000000 },
+    [SBSA_GIC_ITS] =            { 0x44090000, 0x00020000 },
     [SBSA_SECURE_EC] =          { 0x50000000, 0x00001000 },
     [SBSA_GWDT_REFRESH] =       { 0x50010000, 0x00001000 },
     [SBSA_GWDT_CONTROL] =       { 0x50011000, 0x00001000 },
@@ -392,7 +394,29 @@ static void create_secure_ram(SBSAMachineState *sms,
     memory_region_add_subregion(secure_sysmem, base, secram);
 }
 
-static void create_gic(SBSAMachineState *sms)
+static void create_its(SBSAMachineState *sms)
+{
+    const char *itsclass = its_class_name();
+    DeviceState *dev;
+
+    if (!strcmp(itsclass, "arm-gicv3-its")) {
+        itsclass = NULL;
+    }
+
+    if (!itsclass) {
+        /* Do nothing if not supported */
+        return;
+    }
+
+    dev = qdev_new(itsclass);
+
+    object_property_set_link(OBJECT(dev), "parent-gicv3", OBJECT(sms->gic),
+                             &error_abort);
+    sysbus_realize_and_unref(SYS_BUS_DEVICE(dev), &error_fatal);
+    sysbus_mmio_map(SYS_BUS_DEVICE(dev), 0, sbsa_ref_memmap[SBSA_GIC_ITS].base);
+}
+
+static void create_gic(SBSAMachineState *sms, MemoryRegion *mem)
 {
     unsigned int smp_cpus = MACHINE(sms)->smp.cpus;
     SysBusDevice *gicbusdev;
@@ -419,6 +443,12 @@ static void create_gic(SBSAMachineState *sms)
     qdev_prop_set_uint32(sms->gic, "len-redist-region-count", 1);
     qdev_prop_set_uint32(sms->gic, "redist-region-count[0]", redist0_count);
 
+    if (!kvm_irqchip_in_kernel()) {
+        object_property_set_link(OBJECT(sms->gic), "sysmem",
+                                OBJECT(mem), &error_fatal);
+        qdev_prop_set_bit(sms->gic, "has-lpi", true);
+    }
+
     gicbusdev = SYS_BUS_DEVICE(sms->gic);
     sysbus_realize_and_unref(gicbusdev, &error_fatal);
     sysbus_mmio_map(gicbusdev, 0, sbsa_ref_memmap[SBSA_GIC_DIST].base);
@@ -465,6 +495,7 @@ static void create_gic(SBSAMachineState *sms)
         sysbus_connect_irq(gicbusdev, i + 3 * smp_cpus,
                            qdev_get_gpio_in(cpudev, ARM_CPU_VFIQ));
     }
+    create_its(sms);
 }
 
 static void create_uart(const SBSAMachineState *sms, int uart,
@@ -770,7 +801,7 @@ static void sbsa_ref_init(MachineState *machine)
 
     create_secure_ram(sms, secure_sysmem);
 
-    create_gic(sms);
+    create_gic(sms, sysmem);
 
     create_uart(sms, SBSA_UART, sysmem, serial_hd(0));
     create_uart(sms, SBSA_SECURE_UART, secure_sysmem, serial_hd(1));
```

chetan-rathore, 2022-12-13:
Hi @shashi-j, 

Thanks for the steps on enabling ITS, we are checking it on our side.

For the PCIe-related discussion, can you share 
1) pci cmd dump. It will show all the pcie devices on the system.
2) For each pcie device listed as output of previous cmd, please run  " pci bus dev func  -i " cmd and share its dump....it will help in knowing the device class code.
3) Run "Bsa.efi -v 1 -mmio" and share its output.  This will help in knowing which all devices where identified by BSA and why were they omitted.  Please note it will take some time (On the qemu at our side , it took around 10-15 minutes)

shashi-j, 2022-12-13:
Hi Chetan,

Please find attached the output logs file  
[pci_cmds.log](https://github.com/ARM-software/bsa-acs/files/10220194/pci_cmds.log) for commands requested

chetan-rathore, 2022-12-13:
Hi @shashi-j, 

From the logs, I can see the Ethernet ctrl and Serial Bus ctrl are RCiEP  and Display Controller is not an PCI express device.
So BSA will ignore all the 3 devices.

1) 00 01 00   device
 Class: Network Controller - Ethernet controller - 
Pci Express device capability structure:
CapID( 0):          10            NextCap Ptr( 1):    00
Cap Register( 2):             0091
  Capability Version(3:0):          0x0001
  Device/PortType(7:4):             Root Complex Integrated Endpoint

2) 00 02 00  device
Class: Display Controller - VGA/8514 controller - VGA compatible
// No express capabilities

3) 00 03 00 device
Class: Serial Bus Controllers - USB - xHCI

Pci Express device capability structure:
CapID( 0):          10            NextCap Ptr( 1):    00
Cap Register( 2):             0092
  Capability Version(3:0):          0x0002
  Device/PortType(7:4):             Root Complex Integrated Endpoint

shashi-j, 2022-12-13:
Hi Chetan,

I had also tried by adding a nic pci device as below,but that showed the same behaviour too.Even in sbsa.efi running test case number -t 441 didnt happen,showed the same behavior.

```
FS0:\> pci                 
   Seg  Bus  Dev  Func
   ---  ---  ---  ----
    00   00   00    00 ==> Bridge Device - Host/PCI bridge
             Vendor 1B36 Device 0008 Prog Interface 0
    00   00   01    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   00   02    00 ==> Display Controller - VGA/8514 controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   00   03    00 ==> Serial Bus Controllers - USB
             Vendor 1B36 Device 000D Prog Interface 30
    00   00   04    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 100E Prog Interface 0
FS0:\> pci 00 04 00 -i     
  PCI Segment 00 Bus 00 Device 04 Func 00 [EFI 0000040000]
  00000000: 86 80 0E 10 00 00 00 00-03 00 00 02 00 00 00 00  *................*
  00000010: 00 00 04 81 01 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000020: 00 00 00 00 00 00 00 00-00 00 00 00 F4 1A 00 11  *................*
  00000030: 00 00 FC FF 00 00 00 00-00 00 00 00 FF 01 00 00  *................*

  00000040: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000050: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000060: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000070: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000080: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000090: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000A0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000B0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000C0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000E0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000F0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*

Vendor ID(0): 8086     Device ID(2): 100E
Command(4): 0000
  (00)I/O space access enabled:       0  (01)Memory space access enabled:    0
  (02)Behave as bus master:           0  (03)Monitor special cycle enabled:  0
  (04)Mem Write & Invalidate enabled: 0  (05)Palette snooping is enabled:    0
  (06)Assert PERR# when parity error: 0  (07)Do address/data stepping:       0
  (08)SERR# driver enabled:           0  (09)Fast back-to-back transact...:  0

Status(6): 0000
  (04)New Capabilities linked list:   0  (05)66MHz Capable:                  0
  (07)Fast Back-to-Back Capable:      0  (08)Master Data Parity Error:       0
  (09)DEVSEL timing:               Fast  (11)Signaled Target Abort:          0
  (12)Received Target Abort:          0  (13)Received Master Abort:          0
  (14)Signaled System Error:          0  (15)Detected Parity Error:          0

Revision ID(8): 03 BIST(0F):  Incapable
Cache Line Size(C): 00                 Latency Timer(D): 00
Header Type(0E):    00, Single function, PCI device
Class: Network Controller - Ethernet controller - 
FS0:\> 
```

chetan-rathore, 2022-12-14:
Hi @shashi-j , 

Based on pci output of Ethernet Ctrl shared, the device doesn't have express capabilities. It is a PCI device and not PCIe.

The dump of ethernet ctrl at my side is.

FS0:\> pci
   Seg  Bus  Dev  Func
   ---  ---  ---  ----
    00   00   00    00 ==> Bridge Device - Host/PCI bridge
             Vendor 1B36 Device 0008 Prog Interface 0
    00   00   01    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   00   02    00 ==> Display Controller - VGA/8514 controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   00   03    00 ==> Serial Bus Controllers - USB
             Vendor 1B36 Device 000D Prog Interface 30

FS0:\> pci 00 01 00 -i
  PCI Segment 00 Bus 00 Device 01 Func 00 [EFI 0000010000]
  00000000: 86 80 D3 10 00 00 10 00-00 00 00 02 00 00 00 00  *................*
  00000010: 00 00 06 81 00 00 04 81-01 00 00 00 00 00 08 81  *................*
  00000020: 00 00 00 00 00 00 00 00-00 00 00 00 86 80 00 00  *................*
  00000030: 00 00 FC FF C8 00 00 00-00 00 00 00 FF 01 00 00  *................*

  00000040: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000050: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000060: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000070: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000080: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000090: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000A0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000B0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000C0: 00 00 00 00 00 00 00 00-01 E0 22 00 00 00 00 00  *..........".....*
  000000D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000000E0: 10 00 91 00 00 80 00 00-00 00 00 00 11 04 00 00  *................*
  000000F0: 00 00 11 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*

Start dumping PCIex extended configuration space (0x100 - 0xFFF).

  00000100: 01 00 02 14 00 00 00 00-00 00 00 00 30 20 46 00  *............0 F.*
  00000110: 00 00 00 00 00 E0 00 00-A0 00 00 00 00 00 00 00  *................*
  00000120: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000130: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000140: 03 00 01 00 56 34 12 FF-FF 00 54 52 00 00 00 00  *....V4....TR....*
  00000150: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000160: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000170: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000180: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000190: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000001A0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000001B0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000001C0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000001D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000001E0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000001F0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000200: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000210: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000220: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000230: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000240: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000250: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000260: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000270: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000280: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  00000290: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000002A0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000002B0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000002C0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
  000002D0: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................*
[... 279 lines cut ...]
    AdvancedErrorCapAndControl     000000A0
    HeaderLog1                     00000000
    HeaderLog2                     00000000
    HeaderLog3                     00000000
    HeaderLog4                     00000000
    RootErrorCommand               00000000
    RootErrorStatus                00000000
    ErrorSourceIdentification      0000
    CorrectableErrorSourceIden     0000
    TlpPrefixLog1                  00000000
    TlpPrefixLog2                  00000000
    TlpPrefixLog3                  00010003
    TlpPrefixLog4                  FF123456
    00000100: 01 00 02 14 00 00 00 00-00 00 00 00 30 20 46 00  *............0 F.
*
    00000110: 00 00 00 00 00 E0 00 00-A0 00 00 00 00 00 00 00  *................
*
    00000120: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................
*
    00000130: 00 00 00 00 00 00 00 00-00 00 00 00 00 00 00 00  *................
*
    00000140: 03 00 01 00 56 34 12 FF-                         *....V4..*
  Serial Number
    SerialNumber                   525400FFFF123456
    00000140: 03 00 01 00 56 34 12 FF-FF 00 54 52              *....V4....TR*

chetan-rathore, 2022-12-16:
Hi @shashi-j, 

We tried with enabling ITS  steps at our end and same exception is encountered.  
But the exception is not due to ACS, as directly accessing the ITS base (0x44090000) from the uefi shell.
mm -w 4 0x44090000   is also resulting into exception. 

It will seem that some more changes during building QEMU might be require so that the address can be accessed.

Thanks, 
ACS team

shashi-j, 2022-12-16:
Hi Chetan,

Will look into this exception.
On a generic note,do you have a log of ACS tests running an individual test case for any other platform?

chetan-rathore, 2022-12-17:
Hi Shashi, 

Please find attached latest SBSA acs logs on QEMU (sbsa-ref) machine at our side.

Command used:  Sbsa.efi -t 441 -v 1 -l 6

[log.txt](https://github.com/ARM-software/bsa-acs/files/10251374/log.txt)

shashi-j, 2022-12-19:
Hi Chetan,

From the log,i see that -t 441 is specified explicitly ,but since you are running level 6 ,all tests (including 441) would get executed anyway.So,what difference does "-t 441" make? I was of the impression that with -t <test-case-num>,only that test case should run skipping all others from the level.Please clarify.

chetan-rathore, 2022-12-27:
Hi Shashi, 

-t option is used to run a single test, but the level at which that test should run will be applicable and need to be passed.

shashi-j, 2023-01-06:
Hi Chetan,

I got the SBSA test case 441 to work by implementing the missing IORT table in edk2.The test case is passing now (log below)

```
441 : Check MSI and MSI-X support rule  
       START  
    BDF 0x100 int pin value 1 MSI cap 0 MSIX cap 0
    BDF 0x300 int pin value 1 MSI cap 268435472 MSIX cap 0: Result:  PASS 
       END 
``` 
Closing this issue now.

closed 2023-01-06 by shashi-j

chetan-rathore, 2023-01-07:
Great @shashi-j, 

Is the IORT table change is now upstreamed to edk2 github or in some of your local fork.
We will also like to setup our test enviroment with latest changes.

shashi-j, 2023-01-09:
The IORT table change hasn't been upstreamed yet.It is planned to be done part overall ITS upstream.
Right now these changes are local

chetan-rathore, 2023-01-10:
Okay..thanks for update