ARM-software/bsa-acs #80: Issue with test_os_its003.c

tvidyasagar, 2022-12-19:
Hi,
Looking at the code in test_os_its003.c, it seems to assume that there is only one SMMU in the system. If we have a system like the one attached in the pic where we have multiple SMMU instances and PCIe RCs behind them having same StreamIDs, then, uniqueness of streamID should always be checked in the context of the specific SMMU instance i.e. to put it differently, in the context of a specific output reference.
[image]

rajatgoyal47, 2023-01-11:
Hi Vidya,

Thank you for reporting this.
We will fix this and upstream the changes soon. 

Thanks,
Rajat Goyal
ACS Team

rajatgoyal47, 2023-01-12:
Hi Vidya,

Fix has been merged as part of #81
Thanks for reporting this.

Rajat Goyal
ACS Team

chetan-rathore, 2023-01-23:
Hi @tvidyasagar, 

Closing the issue now. Please reopen if the issue is seen.

Thanks,
ACS team

closed 2023-01-23 by chetan-rathore