ARM-software/bsa-acs #85: test_os_m002.c is not buildable

hrw, 2023-01-12:
```
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/test_pool/memory_map/operating_system/test_os_m002.c: In function ‘payload’:
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/test_pool/memory_map/operating_system/test_os_m002.c:89:18: error: storing the address of local variable ‘exception_taken_n’ in ‘branch_to_test’ [-Werror=dangling-pointer=]
   89 |   branch_to_test = &&exception_taken_n;
      |   ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/test_pool/memory_map/operating_system/test_os_m002.c:106:1: note: ‘exception_taken_n’ declared here
  106 | exception_taken_n:
      | ^~~~~~~~~~~~~~~~~
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/test_pool/memory_map/operating_system/test_os_m002.c:31:14: note: ‘branch_to_test’ declared here
   31 | static void *branch_to_test;
      |              ^~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make[1]: *** [GNUmakefile:991: /home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/bsa-acs/uefi_app/BsaAcs/OUTPUT/test_pool/memory_map/operating_system/test_os_m002.obj] Błąd 1
make[1]: Opuszczenie katalogu '/home/marcin/devel/linaro/sbsa-qemu/code/Build/Shell/DEBUG_GCC49/AARCH64/ShellPkg/Application/bsa-acs/uefi_app/BsaAcs'
```

```
18:19 marcin@puchatek:code$ aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (GCC) 12.2.1 20220819 (Red Hat Cross 12.2.1-1)
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

hrw, 2023-01-12:
Something like https://github.com/ARM-software/sbsa-acs/pull/285 again?

chetan-rathore, 2023-01-23:
Hi @hrw,

The changes for memory test 2 (m002) are merged as part of the recent BSA v1.0.3 release.
Can you please confirm if it's getting compiled at your end also.

Thanks,
ACS team

hrw, 2023-01-23:
I wonder will there be a day when your code drops will be split into sensible commits.

ACS feels like proprietary project with source under open source license. With random code drops from time to time without any split of changes into sensible parts.

closed 2023-01-23 by hrw