ARM-software/bsa-acs #86: 4KB Granule support test is wrong

hrw, 2023-01-12:
On QEMU with 'max' we have interesting bug:

```
   4 : Check PE 4KB Granule Support
       Failed on PE -    0
       Checkpoint --  1                           : Result:  FAIL
```

Test checks TGran4 bits [31:28] in ID_AA64MMFR0_EL1 for value 0 (which means "4KB granule supported").

We have 0x0001 there which means "4KB granule supports 52-bit input and output addresses" ([Arm arm](https://developer.arm.com/documentation/ddi0601/2022-03/AArch64-Registers/ID-AA64MMFR0-EL1--AArch64-Memory-Model-Feature-Register-0?lang=en#fieldset_0-31_28)).

chetan-rathore, 2023-01-23:
Hi @hrw, 

Thanks for raising this.
The test will need an update to take into FEAT_LPA2 implementation by PE.

Thanks,
ACS team

chetan-rathore, 2023-01-27:
Hi @hrw, 

Changes for the issue are merged.

Thanks,
ACS team

hrw, 2023-01-27:
```
   4 : Check PE 4KB Granule Support               : Result:  PASS
```

closed 2023-01-27 by hrw