ARM-software/bsa-acs #87: Do not inform that system does not have GICv2m

hrw, 2023-01-12:
Can we agree that GICv2m is distant past and just do not print that we skip tests related to it?

Now on GICv3+ system:

```
      ***  Starting GIC tests ***
Operating System View:
 201 : Check GIC version                          : Result:  PASS
 202 : Check GICv2 Valid Configuration
       Checkpoint --  1                           : Result:  SKIPPED
 203 : If PCIe, GICv3 then ITS, LPI               : Result:  PASS
 204 : Check GICv3 Security States                : Result:  PASS
 205 : Non-secure SGIs are implemented            : Result:  PASS
 206 : Check PPI Assignments for OS               : Result:  PASS

Hypervisor View:
 226 : Check PPI Assignments Hypervisor
       EL2-Virtual timer interrupt 28 not received     : Result:  PASS

       No GICv2m, Skipping all GICv2m tests

      *** Starting ITS tests ***
```

Cleaner output on GICv3+ system:

```
      ***  Starting GIC tests ***
Operating System View:
 201 : Check GIC version                          : Result:  PASS
 203 : If PCIe, GICv3 then ITS, LPI               : Result:  PASS
 204 : Check GICv3 Security States                : Result:  PASS
 205 : Non-secure SGIs are implemented            : Result:  PASS
 206 : Check PPI Assignments for OS               : Result:  PASS

Hypervisor View:
 226 : Check PPI Assignments Hypervisor
       EL2-Virtual timer interrupt 28 not received     : Result:  PASS

      *** Starting ITS tests ***
```

hrw, 2023-01-12:
Not to mention that output of test 226 would need some alignment too.

chetan-rathore, 2023-01-23:
Hi @hrw , 

As the BSA specification allows GICv2, v2m as valid configurations, the test suite logs will have prints related to them.
But we agree that if system doesn't have v2 or v2m, then the test can directly be skipped from running.

We will make the required changes.

Thanks,
ACS team

hrw, 2023-01-23:
I meant more of this:

```
1. BSA starts
2. checks is there GIC in system
3. runs GIC tests
4. test 201 starts and since then we know which GIC version is present
5. if GIC >= 3 then
   1. skip 202
   2. run 203 etc
   3. skip GIC v2m tests
6. run ITS tests
```

chetan-rathore, 2023-01-27:
Hi @hrw, 

Code changes are made to 
1) Skip 202 for GICv3 system
2) Moved  " No GICv2m, Skipping all GICv2m tests"  from TEST to higher verbosity prints.

Also, an additional print to log which GIC version is detected in the system is added.
The print will be in below format.

GIC INFO: GIC version : v3
OR
GIC INFO: GIC version : v2m
OR
GIC INFO: GIC version : v2

Thanks,
ACS team.

chetan-rathore, 2023-01-27:
We are checking the 226 test alignment issue, and update to it will be made along with the PR if #94

PR #98 (2023-01-27): Run test 202 (B_GIC_02) only for system with GICv2
 - changed verbosity of GICv2m skip print to INFO.
 - checks added to run test 202 only if GICv2 is present.
 - added GIC version print.

Fixes #87
closed 2023-01-27 by chetan-rathore