ARM-software/bsa-acs #90: StreamID related tests (278 and 279) should not be run if the SoC doens't have SMMU

sunnywang-arm, 2023-01-18:
Now the issue causes false-positive on the systems that don't have SMMU. 
We might need to move StreamID-related tests from ITS tests to SMMU tests.

hrw, 2023-01-25:
sbsa-ref gives this:

```
      *** Starting ITS tests ***

Operating System View:
 276 : Check number of ITS blocks in a group      : Result:  PASS
 277 : Check ITS block association with group     : Result:  PASS
 278 : Check uniqueness of StreamID
       Failed on PE -    0
       Checkpoint --  3                           : Result:  FAIL
 279 : Check Device's ReqID-DeviceID-StreamID
       Failed on PE -    0
       Checkpoint --  3                           : Result:  FAIL

      *** One or more tests have Failed/Skipped.***

      *** Starting System MMU tests ***
       No SMMU Controller Found, Skipping SMMU tests...
```

(ITS patch is not upstreamed yet)

hrw, 2023-01-25:
And with "-v 1" it is even funnier:

```
 278 : Check uniqueness of StreamID                   START
 No SMMU found behind the RootComplex with seg :0RC with segment number 0 is not
 behind any SMMU No SMMU found behind the RootComplex with seg :0RC with segment
 number 0 is not behind any SMMU No SMMU found behind the RootComplex with seg :
0RC with segment number 0 is not behind any SMMU No SMMU found behind the RootCo
mplex with seg :0RC with segment number 0 is not behind any SMMU No SMMU found b
ehind the RootComplex with seg :0RC with segment number 0 is not behind any SMMU
 No SMMU found behind the RootComplex with seg :0RC with segment number 0 is not
 behind any SMMU No SMMU found behind the RootComplex with seg :0RC with segment
 number 0 is not behind any SMMU No SMMU found behind the RootComplex with seg :
0RC with segment number 0 is not behind any SMMU No SMMU found behind the RootCo
mplex with seg :0RC with segment number 0 is not behind any SMMU
       Stream ID is not unique for bdf : 0x50000
       Failed on PE -    0
       ITS_DEV_2
       Checkpoint --  3                           : Result:  FAIL
         END

 279 : Check Device's ReqID-DeviceID-StreamID         START

       ReqID-StreamID Association Fail for Bdf : 400
       Failed on PE -    0
       ITS_DEV_7,ITS_DEV_8
       Checkpoint --  3                           : Result:  FAIL
         END
```

rajatgoyal47, 2023-01-27:
Hi Sunny/Marcin,

Thank you for reporting this.

Test 278 : StreamID check only test, this will skip in the systems where SMMU is not present.
Test 279 : There are two checks in this tests, One check in case when SMMU is present where it checks for DeviceID-StreamID-ReqID relation. Other when SMMU is not present, in this case it will check for DeviceID-ReqID relation. This test will be run.

We are making the changes and will be upstreamed soon.

Thanks,
Rajat 
ACS team

rajatgoyal47, 2023-01-30:
Hi Sunny/Marcin,

Changes are merged as part of #100.

For the -v 1 & -v 2 logs with higher verbosity, we are making changes to improve the logs which will be part of different pull request.

Thanks,
Rajat 
ACS Team

hrw, 2023-01-30:
Updated to latest (commit 47c89092112d48dc415e31fc96bd55b7c274a0d5) git:

```
 BSA Architecture Compliance Suite
[..]
 SMMU_INFO: Number of SMMU CTRL       :    0
[..]
      *** Starting DeviceID generation and ITS tests ***

Operating System View:
 276 : Check number of ITS blocks in a group      : Result:  PASS
 277 : Check ITS block association with group     : Result:  PASS
 278 : Check uniqueness of StreamID
       Checkpoint --  2                           : Result:  SKIPPED
 279 : Check Device's ReqID-DeviceID-StreamID     : Result:  PASS

      *** One or more tests have Failed/Skipped.***

      *** Starting System MMU tests ***
       No SMMU Controller Found, Skipping SMMU tests...
```

hrw, 2023-01-30:
With "-v 1":

```
 279 : Check Device's ReqID-DeviceID-StreamID         START
 No SMMU found behind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x300 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x400 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x500 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x600 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x700 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x10000 Not Behind an SMMU, Skipping StreamID check No SMMU found b
ehind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x20000 Not Behind an SMMU, Skipping StreamID check No SMMU found b
ehind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x40000 Not Behind an SMMU, Skipping StreamID check No SMMU found b
ehind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x50000 Not Behind an SMMU, Skipping StreamID check
       ITS_DEV_7,ITS_DEV_8
                                       : Result:  PASS 
         END```

sunnywang-arm, 2023-01-30:
Thanks for testing the fix, @hrw.  
I also verified the fix on my system. It works fine, so let's close the issue.

hrw, 2023-01-30:
Looking at output of test 279 on smmuless system I would not call it fixed.

sunnywang-arm, 2023-01-31:
Reopen issue because of comment from @hrw 
@hrw  Do you mean the result of 279 should show SKIPPED instead of PASS? If not, Could you elaborate on the issue/problem with 279?

hrw, 2023-01-31:
Test 278 tests for uniqueness of StreamID and gets SKIPPED due to lack of SMMU:

```
 278 : Check uniqueness of StreamID                   START

       No SMMU, Skipping Test
       ITS_DEV_2
       Checkpoint --  2                           : Result:  SKIPPED
         END
```

And then test 279 happily announce lack of SMMU and that it skips doing check:

```
 No SMMU found behind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x300 Not Behind an SMMU, Skipping StreamID check
```

And then announce PASS:

```
 279 : Check Device's ReqID-DeviceID-StreamID         START
 No SMMU found behind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x300 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x400 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x500 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x600 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x700 Not Behind an SMMU, Skipping StreamID check No SMMU found beh
ind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x10000 Not Behind an SMMU, Skipping StreamID check No SMMU found b
ehind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x20000 Not Behind an SMMU, Skipping StreamID check No SMMU found b
ehind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x40000 Not Behind an SMMU, Skipping StreamID check No SMMU found b
ehind the RootComplex with seg :0
       RC with segment number 0 is not behind SMMU
       BDF : 0x50000 Not Behind an SMMU, Skipping StreamID check
       ITS_DEV_7,ITS_DEV_8
                                       : Result:  PASS
         END
```

Both tests skip doing work due to lack of SMMU so both should have result SKIPPED.

hrw, 2023-01-31:
Also note lack of "\n" between "Skipping StreamID check" and "No SMMU found".

rajatgoyal47, 2023-02-01:
Hi Marcin,

Bellow are the expected results in this case.

Test 278 : This is a StreamID check only test, this will skip in the systems where SMMU is not present. Skipping as expected.
 
Test 279 : There are two checks in this tests, One check in case when SMMU is present where it checks for DeviceID-StreamID-ReqID relation. Other when SMMU is not present, in this case it will check for DeviceID-ReqID relation.

In current scenario, First StreamID related check is skipping, hence the prints in -v 1 case. and the 2nd Check is being done, that's why test is showing as passed.
Regarding the \n issue, we will take care of this in separate code change.

Please let me know if you have any other feedback/observations.

Thanks,
Rajat

hrw, 2023-02-01:
If test 279 has two different checks then maybe consider splitting it?

So system vendor can see 279a failing due to lack of smmu and 279b passing. Then once smmu gets added next run will show 'o, you have smmu but 279a still fails, work on it'.

sunnywang-arm, 2023-02-01:
If we have a reason to put 279a and 279b together, maybe showing some messages related to 279b could be the solution. Now the confusing part is that all the messages from 279 are showing the test being skipped. No message shows that the 279 still tests something for the case where SMMU is not present.

rajatgoyal47, 2023-02-01:
Test has two checks which are related and part of a single rule as per the spec. Issue here is we print messages in case of Skip/Fail hence we are seeing the messages for StreamID check, for the 2nd check everything is passing so we don't see any messages.

We will improve the -v 1 logs to properly show the checks which are being done.

chetan-rathore, 2023-03-14:
Hi @hrw,

Changes are done to improve debug prints of the tests.

Thanks,
ACS team