ARM-software/bsa-acs #91: PCIE_INFO: Number of BDFs found      :    A

hrw, 2023-01-23:
I was playing with sbsa-ref with more complex PCI Express setup and BSA ACS gave me weird output:

```
 PCIE_INFO: Number of ECAM regions    :    1 
  Dev bdf 0x300 type 0x10 RP bdf 0x300
  Dev bdf 0x400 type 0x10 RP bdf 0x400
  Dev bdf 0x500 type 0x10 RP bdf 0x500
  Dev bdf 0x600 type 0x10 RP bdf 0x600
  Dev bdf 0x700 type 0x10 RP bdf 0x700
  Dev bdf 0x10000 type 0x1 RP bdf 0x300
  Dev bdf 0x20000 type 0x1 RP bdf 0x400
  Dev bdf 0x30000 type 0x1 RP bdf 0x500
  Dev bdf 0x40000 type 0x1 RP bdf 0x600
  Dev bdf 0x50000 type 0x80 RP bdf 0x700
 PCIE_INFO: Number of BDFs found      :    A
```

I would expect "10" in this place.

```
Shell> pci
   Seg  Bus  Dev  Func
   ---  ---  ---  ----
    00   00   00    00 ==> Bridge Device - Host/PCI bridge
             Vendor 1B36 Device 0008 Prog Interface 0
    00   00   01    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   00   02    00 ==> Display Controller - VGA/8514 controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   00   03    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   04    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   05    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   06    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   07    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   01   00    00 ==> Mass Storage Controller - Non-volatile memory subsystem
             Vendor 1B36 Device 0010 Prog Interface 2
    00   02   00    00 ==> Serial Bus Controllers - USB
             Vendor 1B36 Device 000D Prog Interface 30
    00   03   00    00 ==> Display Controller - Other display controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   04   00    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   05   00    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000E Prog Interface 0
    00   06   08    00 ==> Pre 2.0 device - UNDEFINED
             Vendor 1AF4 Device 1005 Prog Interface 0
Shell> 
```

```
~ # lspci -t
-[0000:00]-+-00.0
           +-01.0
           +-02.0
           +-03.0-[01]----00.0
           +-04.0-[02]----00.0
           +-05.0-[03]----00.0
           +-06.0-[04]----00.0
           \-07.0-[05-06]----00.0-[06]----08.0
~ # lspci -nn
00:00.0 Host bridge [0600]: Red Hat, Inc. QEMU PCIe Host bridge [1b36:0008]
00:01.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
00:02.0 VGA compatible controller [0300]: Device [1234:1111] (rev 02)
00:03.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
00:04.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
00:05.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
00:06.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
00:07.0 PCI bridge [0604]: Red Hat, Inc. QEMU PCIe Root port [1b36:000c]
01:00.0 Non-Volatile memory controller [0108]: Red Hat, Inc. QEMU NVM Express Controller [1b36:0010] (rev 02)
02:00.0 USB controller [0c03]: Red Hat, Inc. QEMU XHCI Host Controller [1b36:000d] (rev 01)
03:00.0 Display controller [0380]: Device [1234:1111] (rev 02)
04:00.0 Ethernet controller [0200]: Intel Corporation 82574L Gigabit Network Connection [8086:10d3]
05:00.0 PCI bridge [0604]: Red Hat, Inc. Device [1b36:000e]
06:08.0 Unclassified device [00ff]: Red Hat, Inc. Virtio RNG [1af4:1005]
```

hrw, 2023-01-23:
Command line:

```
./code/qemu/build/aarch64-softmmu/qemu-system-aarch64 \
-machine sbsa-ref \
-m 4096 \
-smp 2 \
-cpu max \
\
-drive if=pflash,file=SBSA_FLASH0.fd,format=raw \
-drive if=pflash,file=SBSA_FLASH1.fd,format=raw \
\
-device pcie-root-port,id=root_port1,chassis=1,slot=1  \
  -device nvme,serial=deadbeef,drive=nvme,bus=root_port1 \
-device pcie-root-port,id=root_port2,chassis=1,slot=2  \
  -device qemu-xhci,id=usb,bus=root_port2 \
-device pcie-root-port,id=root_port3,chassis=1,slot=3  \
  -device bochs-display,bus=root_port3 \
-device pcie-root-port,id=root_port4,chassis=1,slot=4  \
  -device e1000e,bus=root_port4 \
-device pcie-root-port,id=root_port5,chassis=1,slot=5  \
  -device pcie-pci-bridge,id=pci,bus=root_port5 \
    -device virtio-rng-pci,bus=pci,addr=8 \
\
-drive file=fat:rw:$PWD/disks/virtual/,format=raw \
-drive if=ide,format=raw,media=cdrom,file=disks/debian-bookworm-DI-alpha1-arm64-netinst.iso \
-drive file=disks/efi-tools.img,if=none,id=nvme,format=raw \
\
-usb \
-device usb-kbd \
-device usb-tablet \
\
-watchdog-action none \
-no-reboot \
-monitor telnet::45454,server,nowait \
-serial stdio
```

chetan-rathore, 2023-01-27:
Hi @hrw, 

Thanks for highlighting it.
All the INFO prints will output in decimal format.

Thanks,
ACS team

chetan-rathore, 2023-01-27:
Hi @hrw, 

Thanks for highlighting the issue. Changes are merged.

Thanks,
ACS team

hrw, 2023-01-30:
``` PCIE_INFO: Number of BDFs found      :    10```

closed 2023-01-30 by hrw