ARM-software/bsa-acs #107: Improvements for PCIe tests' message

sunnywang-arm, 2023-02-14:
Some PCIe tests' output only has multiple "DP type  1" messages like the following, which is not meaningful. 
```
 803 : All EP/Sw under RP in same ECAM Region         START
 DP type  1 
 DP type  1 
 DP type  1 
 DP type  1 
 DP type  1 
 DP type  1 
 DP type  1 
```
Therefore, it looks like we can do some improvements like the following:  
- For print level 3 (default level), it might be good to let the test cases not print out the message "DP type  1".  Only print "DP type  1" with level 1. 
- For print level 1 (-v 1), It might be good to let the test case also print BDF info to know what devices have been checked. 

Besides 803, I also saw this problem with the tests 804, 805, 809, 811, 817, 821, 822, 824, 825, 835, 836, 837, 838, 839.

For tests 831, 832, 833, 836, 837, 838, it looks like we don’t have messages showing what devices have been checked. We might want to add the message for this.

chetan-rathore, 2023-02-20:
Thanks, @sunnywang-arm for the improvement. Debug prints will be added to show which devices are been checked.

chetan-rathore, 2023-03-13:
Hi @sunnywang-arm, 

Closing the issue as changes are merged.

closed 2023-03-13 by chetan-rathore