ARM-software/bsa-acs #117: List SMMU version in information block

hrw, 2023-03-03:
There is GIC version listed at start of app. Would be nice to have SMMU version there as well:

```
 Creating Platform Information Tables
 PE_INFO: Number of PE detected       :    4 
 GIC INFO: GIC version                :    v3 
 GIC_INFO: Number of GICD             :    1 
 GIC_INFO: Number of GICR RD          :    1 
 GIC_INFO: Number of ITS              :    1 
 TIMER_INFO: Number of system timers  :    0 
 WATCHDOG_INFO: Number of Watchdogs   :    1 
 PCIE_INFO: Number of ECAM regions    :    1 
 PCIE_INFO: Number of BDFs found      :    10
 PCIE_INFO: Number of RCiEP           :    0 
 PCIE_INFO: Number of RCEC            :    0 
 PCIE_INFO: Number of EP              :    4 
 PCIE_INFO: Number of RP              :    5 
 PCIE_INFO: Number of iEP_EP          :    0 
 PCIE_INFO: Number of iEP_RP          :    0 
 SMMU_INFO: Number of SMMU CTRL       :    1 
 Peripheral: Num of USB controllers   :    1 
 Peripheral: Num of SATA controllers  :    1 
 Peripheral: Num of UART controllers  :    1 

```

chetan-rathore, 2023-03-06:
Hi @hrw, 

Thanks for the suggestion. We will add the required print.

Thanks,
ACS team

hrw, 2023-03-09:
done

hrw, 2023-03-10:
```
 SMMU_INFO: SMMU index 00 version     :    v3 
```

It does not tell is it v3.0 or v3.2 which make a difference for test 701.

hrw, 2023-03-10:
Verbose output also lacks such info:

```
 Creating SMMU INFO table
  IORT node offset:30, type: 0
  IORT node offset:48, type: 4
  IORT node offset:30, type: 0
  IORT node offset:28, type: 48
 Invalid IORT node type
  IORT node offset:A0, type: 2
  IORT node offset:48, type: 4
  Number of IOVIRT blocks = 3
  ITS Group Num ITS: 1
  ITS ID: 0
  SMMU: Major Rev:3 Base Address:0x60050000
  Number of ID Mappings:2
  input_base:0x0 id_count:0xFFFF
  output_base:0x0 output ref:0x18
  input_base:0x3C02 id_count:0x0
  output_base:0x1 output ref:0xFFFFFFFF

  Root Complex  Segment Num:0
  Number of ID Mappings:1
  input_base:0x0 id_count:0xFFFF
  output_base:0x0 output ref:0x134

 SMMU_INFO: Number of SMMU CTRL       :    1 
```

chetan-rathore, 2023-03-14:
Hi @hrw, 

SMMU minor version will also be printed now.

Thanks,
ACS team

hrw, 2023-03-15:
Confirms:

```
 SMMU_INFO: Number of SMMU CTRL       :    1
 SMMU_INFO: SMMU index 00 version     :    v3.1
```

closed 2023-03-15 by hrw