ARM-software/bsa-acs #118: "Invalid IORT node type" written with wrong level

hrw, 2023-03-03:
During printing information table I got "Invalid IORT node type" message:

```
 PCIE_INFO: Number of iEP_EP          :    0 
 PCIE_INFO: Number of iEP_RP          :    0 
 Invalid IORT node type
 SMMU_INFO: Number of SMMU CTRL       :    1 
 Peripheral: Num of USB controllers   :    1 
```

"-v 1" output:

```
 Creating SMMU INFO table
  IORT node offset:30, type: 0
  IORT node offset:48, type: 4
  IORT node offset:30, type: 0
  IORT node offset:28, type: 48
 Invalid IORT node type
  IORT node offset:A0, type: 2
  IORT node offset:48, type: 4
  Number of IOVIRT blocks = 3
  ITS Group Num ITS: 1
  ITS ID: 0
  SMMU: Major Rev:3 Base Address:0x60050000
  Number of ID Mappings:2
  input_base:0x0 id_count:0xFFFF
  output_base:0x0 output ref:0x18
  input_base:0x3C02 id_count:0x0
  output_base:0x1 output ref:0xFFFFFFFF

  Root Complex  Segment Num:0
  Number of ID Mappings:1
  input_base:0x0 id_count:0xFFFF
  output_base:0x0 output ref:0x134

 SMMU_INFO: Number of SMMU CTRL       :    1 
```

I wonder where from it takes those nodes:

```
[000h 0000   4]                    Signature : "IORT"    [IO Remapping Table]
[004h 0004   4]                 Table Length : 000000DC
[008h 0008   1]                     Revision : 00
[009h 0009   1]                     Checksum : F8
[00Ah 0010   6]                       Oem ID : "LINARO"
[010h 0016   8]                 Oem Table ID : "SBSAQEMU"
[018h 0024   4]                 Oem Revision : 20200810
[01Ch 0028   4]              Asl Compiler ID : "LNRO"
[020h 0032   4]        Asl Compiler Revision : 00000001

[024h 0036   4]                   Node Count : 00000003
[028h 0040   4]                  Node Offset : 00000030
[02Ch 0044   4]                     Reserved : 00000000

[030h 0048   1]                         Type : 00
[031h 0049   2]                       Length : 0018
[033h 0051   1]                     Revision : 00
[034h 0052   4]                     Reserved : 00000000
[038h 0056   4]                Mapping Count : 00000000
[03Ch 0060   4]               Mapping Offset : 00000000

[040h 0064   4]                     ItsCount : 00000001
[044h 0068   4]                  Identifiers : 00000000

[048h 0072   1]                         Type : 04
[049h 0073   2]                       Length : 0058
[04Bh 0075   1]                     Revision : 02
[04Ch 0076   4]                     Reserved : 00000000
[050h 0080   4]                Mapping Count : 00000002
[054h 0084   4]               Mapping Offset : 00000044

[058h 0088   8]                 Base Address : 0000000060050000
[060h 0096   4]        Flags (decoded below) : 00000001
                             COHACC Override : 1
                               HTTU Override : 0
                      Proximity Domain Valid : 0
[064h 0100   4]                     Reserved : 00000000
[068h 0104   8]                VATOS Address : 0000000000000000
[070h 0112   4]                        Model : 00000000
[074h 0116   4]                   Event GSIV : 00000000
[078h 0120   4]                     PRI GSIV : 00000000
[07Ch 0124   4]                    GERR GSIV : 00000000
[080h 0128   4]                    Sync GSIV : 00000000
[084h 0132   4]             Proximity Domain : 00000000
[088h 0136   4]      Device ID Mapping Index : 00000001

[08Ch 0140   4]                   Input base : 00000000
[090h 0144   4]                     ID Count : 0000FFFF
[094h 0148   4]                  Output Base : 00000000
[098h 0152   4]             Output Reference : 00000030
[09Ch 0156   4]        Flags (decoded below) : 00000000
                              Single Mapping : 0

[0A0h 0160   4]                   Input base : 00003C02
[0A4h 0164   4]                     ID Count : 00000000
[0A8h 0168   4]                  Output Base : 00000001
[0ACh 0172   4]             Output Reference : 00000028
[0B0h 0176   4]        Flags (decoded below) : 00000001
                              Single Mapping : 1

[0A0h 0160   1]                         Type : 02
[0A1h 0161   2]                       Length : 003C
[0A3h 0163   1]                     Revision : 00
[0A4h 0164   4]                     Reserved : 00000000
[0A8h 0168   4]                Mapping Count : 00000001
[0ACh 0172   4]               Mapping Offset : 00000028

[0B0h 0176   8]            Memory Properties : [IORT Memory Access Properties]
[0B0h 0176   4]              Cache Coherency : 00000001
[0B4h 0180   1]        Hints (decoded below) : 00
                                   Transient : 0
                              Write Allocate : 0
                               Read Allocate : 0
                                    Override : 0
[0B5h 0181   2]                     Reserved : 0000
[0B7h 0183   1] Memory Flags (decoded below) : 00
                                   Coherency : 0
                            Device Attribute : 0
[0B8h 0184   4]                ATS Attribute : 00000001
[0BCh 0188   4]           PCI Segment Number : 00000000
[0C0h 0192   1]            Memory Size Limit : 00
[0C1h 0193   2]           PASID Capabilities : 0000
[0C3h 0195   1]                     Reserved : 00

[0C8h 0200   4]                   Input base : 00000000
[0CCh 0204   4]                     ID Count : 0000FFFF
[0D0h 0208   4]                  Output Base : 00000000
[0D4h 0212   4]             Output Reference : 00000048
[0D8h 0216   4]        Flags (decoded below) : 00000000
                              Single Mapping : 0

Raw Table Data: Length 220 (0xDC)

    0000: 49 4F 52 54 DC 00 00 00 00 F8 4C 49 4E 41 52 4F  // IORT......LINARO
    0010: 53 42 53 41 51 45 4D 55 10 08 20 20 4C 4E 52 4F  // SBSAQEMU..  LNRO
    0020: 01 00 00 00 03 00 00 00 30 00 00 00 00 00 00 00  // ........0.......
    0030: 00 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00  // ................
    0040: 01 00 00 00 00 00 00 00 04 58 00 02 00 00 00 00  // .........X......
    0050: 02 00 00 00 44 00 00 00 00 00 05 60 00 00 00 00  // ....D......`....
    0060: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  // ................
    0070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  // ................
    0080: 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00  // ................
    0090: FF FF 00 00 00 00 00 00 30 00 00 00 00 00 00 00  // ........0.......
    00A0: 02 3C 00 00 00 00 00 00 01 00 00 00 28 00 00 00  // .<..........(...
    00B0: 01 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00  // ................
    00C0: 00 00 00 00 00 00 00 00 00 00 00 00 FF FF 00 00  // ................
    00D0: 00 00 00 00 48 00 00 00 00 00 00 00              // ....H.......
```

hrw, 2023-03-09:
Also done:

```
 PCIE_INFO: Number of iEP_RP          :    0 
 SMMU_INFO: Number of SMMU CTRL       :    1 
 SMMU_INFO: SMMU index 00 version     :    v3 

```

chetan-rathore, 2023-03-09:
hmm...we didn't work explicitly on this ticket, this was the next ticket to be taken up.
@hrw  Is the QEMU firmware used was same in both runs?

We will also check the original issue root cause and which changes fixed it

hrw, 2023-03-09:
My logs show this message now only in "-v 1" log:

https://github.com/hrw/sbsa-ref-status/blob/main/logs/bsa-max.log vs https://github.com/hrw/sbsa-ref-status/blob/main/logs/bsa-max-v1.log

Later today I will add my scripts there.

hrw, 2023-03-09:
I am sorry - I had local change for it.

chetan-rathore, 2023-03-13:
Hi @hrw, 

I had a look on the IORT table, the mapping count for SMMUv3 needs to be 1, as there is only one ITS group.

0050: **02** 00 00 00 44 00 00 00 00 00 05 60 00 00 00 00  // ....D......`....     
[050h 0080   4]                Mapping Count : 00000002

Due to mapping count 2,  the RC node is decoded as second mapping.

[0A0h 0160   4]                   Input base : 00003C02
[0A4h 0164   4]                     ID Count : 00000000
[0A8h 0168   4]                  Output Base : 00000001
[0ACh 0172   4]             Output Reference : 00000028
[0B0h 0176   4]        Flags (decoded below) : 00000001
                              Single Mapping : 1

[0A0h 0160   1]                         Type : 02                           // RC node
[0A1h 0161   2]                       Length : 003C
[0A3h 0163   1]                     Revision : 00
[0A4h 0164   4]                     Reserved : 00000000
[0A8h 0168   4]                Mapping Count : 00000001
[0ACh 0172   4]               Mapping Offset : 00000028

Thanks,
ACS team

chetan-rathore, 2023-03-14:
Thanks @hrw for confirming. 
I will close this issue as no further AI on ACS.

closed 2023-03-14 by chetan-rathore