ARM-software/bsa-acs #128: Add BDF to PCI info

hrw, 2023-03-14:
During "-v 1" run there is PCI info block:

```
 PCIE_INFO: Number of iEP_RP          :    0 
  ECAM 0:  Base 0xF0000000
  Seg: 0x0, Bus: 0x0, Dev: 0x3, Func: 0x0, Dev ID: 0xC, Vendor ID: 0x1B36
  Seg: 0x0, Bus: 0x0, Dev: 0x4, Func: 0x0, Dev ID: 0xC, Vendor ID: 0x1B36
  Seg: 0x0, Bus: 0x0, Dev: 0x5, Func: 0x0, Dev ID: 0xC, Vendor ID: 0x1B36
  Seg: 0x0, Bus: 0x0, Dev: 0x6, Func: 0x0, Dev ID: 0xC, Vendor ID: 0x1B36
  Seg: 0x0, Bus: 0x1, Dev: 0x0, Func: 0x0, Dev ID: 0xD, Vendor ID: 0x1B36
  Seg: 0x0, Bus: 0x2, Dev: 0x0, Func: 0x0, Dev ID: 0x1111, Vendor ID: 0x1234
  Seg: 0x0, Bus: 0x3, Dev: 0x0, Func: 0x0, Dev ID: 0x10D3, Vendor ID: 0x8086
  Seg: 0x0, Bus: 0x4, Dev: 0x0, Func: 0x0, Dev ID: 0xE, Vendor ID: 0x1B36
```

Wherever later PCI card is mentioned it is done by it's BDF value. Can it be added to this table?

```
  bdf: 0x100, Seg: 0x0, Bus: 0x4, Dev: 0x0, Func: 0x0, Dev ID: 0xE, Vendor ID: 0x1B36
```

chetan-rathore, 2023-03-28:
Hi @hrw,
 
ACS team will implement the suggestion.

Thanks,
ACS team

chetan-rathore, 2023-04-10:
Hi @hrw,

Changes for the issue are merged.

Thanks,
ACS team

hrw, 2023-04-11:
```
 Creating PCIe INFO table
  Ecam Index = 0
  Base Address = 0xF0000000
  Segment   = 0x0
  Start Bus = 0x0
  End Bus   = 0xFF
 PCIE_INFO: Number of ECAM regions    :    1 
  Dev bdf 0x000300 type 0x10 RP bdf 0x000300
  Dev bdf 0x000400 type 0x10 RP bdf 0x000400
  Dev bdf 0x000500 type 0x10 RP bdf 0x000500
  Dev bdf 0x000600 type 0x10 RP bdf 0x000600
  Dev bdf 0x010000 type 0x01 RP bdf 0x000300
  Dev bdf 0x020000 type 0x01 RP bdf 0x000400
  Dev bdf 0x030000 type 0x01 RP bdf 0x000500
  Dev bdf 0x040000 type 0x80 RP bdf 0x000600
 PCIE_INFO: Number of BDFs found      :    8
 PCIE_INFO: Number of RCiEP           :    0 
 PCIE_INFO: Number of RCEC            :    0 
 PCIE_INFO: Number of EP              :    3 
 PCIE_INFO: Number of RP              :    4 
 PCIE_INFO: Number of iEP_EP          :    0 
 PCIE_INFO: Number of iEP_RP          :    0 

  ECAM 0:  Base 0xF0000000
  BDF: 0x300
  Seg: 0x0, Bus: 0x00, Dev: 0x03, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x400
  Seg: 0x0, Bus: 0x00, Dev: 0x04, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x500
  Seg: 0x0, Bus: 0x00, Dev: 0x05, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x600
  Seg: 0x0, Bus: 0x00, Dev: 0x06, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x10000
  Seg: 0x0, Bus: 0x01, Dev: 0x00, Func: 0x0, Dev ID: 0x000D, Vendor ID: 0x1B36
  BDF: 0x20000
  Seg: 0x0, Bus: 0x02, Dev: 0x00, Func: 0x0, Dev ID: 0x1111, Vendor ID: 0x1234
  BDF: 0x30000
  Seg: 0x0, Bus: 0x03, Dev: 0x00, Func: 0x0, Dev ID: 0x10D3, Vendor ID: 0x8086
  BDF: 0x40000
  Seg: 0x0, Bus: 0x04, Dev: 0x00, Func: 0x0, Dev ID: 0x000E, Vendor ID: 0x1B36
```

closed 2023-04-11 by hrw