ARM-software/bsa-acs #131: Test 279 getting failed for multiple devices attached to same smmu

abmainkar, 2023-03-22:
In the case of two devices attached to same smmu, smmu_id remains same for both the devices. When test runs for first device, curr_smmu_id gets updated to smmu_id of the first device and curr_grp_sid_cons is calculated. When test runs for second device, smmu_id and curr_smmu_id are same due to which curr_grp_sid_cons remains same 
https://github.com/ARM-software/bsa-acs/blob/main/test_pool/gic/operating_system/test_os_its004.c#L82. 

The error "ReqID-StreamID Association Fail for Bdf : xxxxxxx" gets generated due to if condition https://github.com/ARM-software/bsa-acs/blob/main/test_pool/gic/operating_system/test_os_its004.c#L113

Log for two pcie devices attached to same smmu,

       RC block->data.smmu.base: 15000000
       Seg : 4
       Bus : 0
       Dev : 0
       Func : 0
       its_id : 0
       curr_grp_its_id : 0
       smmu_id : 3
       curr_smmu_id : 2
       stream_id : 10000
       req_id : 0
       device_id : 210000
       curr_grp_sid_cons : 10000
       curr_grp_did_cons : 100000
       
       RC block->data.smmu.base: 15000000
       Seg : 6
       Bus : 0
       Dev : 0
       Func : 0
       its_id  : 0
       curr_grp_its_id  : 0
       smmu_id : 3
       curr_smmu_id : 3
       stream_id : 30000
       req_id : 0
       device_id : 230000
       curr_grp_sid_cons : 10000
       curr_grp_did_cons : 200000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 6000000
       ReqID-StreamID Association Fail for Bdf : 6000000

chetan-rathore, 2023-04-01:
Updates:

In test 279 we are doing ReqID-StreamID check for all the devices behind same SMMU.
 
In this case both the devices are on different Segment but behind the same SMMU, currently test is missing segment number while calculating the req_id hence req_id for both the devices are same with value 0. 
Both the devices have Bus 0, Dev 0, Func 0, but on different segments (Seg 4 & Seg 6).
 
We will update the test to do checks only for devices which are on same segment and behind same SMMU. 

You can access the pre-built binary at this link: 
https://github.com/rajatgoyal47/bsa-acs/blob/main/prebuilt_images/Bsa_fix_279.efi

Please share debug logs with -v 1.

Thanks,
ACS team

abmainkar, 2023-04-03:
Test 279 debug logs with -v 1 after adding patch [https://github.com/rajatgoyal47/bsa-acs/commit/ccae0b8f30dcd6670f68522961f5476f685cb292](https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Frajatgoyal47%2Fbsa-acs%2Fcommit%2Fccae0b8f30dcd6670f68522961f5476f685cb292&data=05%7C01%7Cabmainkar%40nvidia.com%7C56ff6c23ddfa46b8408b08db31a9c25b%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C638158377435581131%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=lKoENtTO63Owb5ScqFDDkULgn%2BcROoFbOKymuD%2BVjTg%3D&reserved=0) 

      279 : Check Device's ReqID-DeviceID-StreamID         START

       RC block->data.smmu.base: 11000000
       RC block->data.smmu.base: 12000000
       RC block->data.smmu.base: 15000000
       RC block->data.smmu.base: 15000000
       RC block->data.smmu.base: 5000000
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8010000
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8010001
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8020000
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8020100
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8020200
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8020300
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8020400
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8020500
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8030000
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8060000
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8080000
       RC block->data.smmu.base: 5000000
       Checking ReqID-StreamID-DeviceID Association, Bdf : 8090000
       RC block->data.smmu.base: 16000000
       ITS_DEV_7,ITS_DEV_8
                                       : Result:  PASS
         END

      All GIC tests passed.

chetan-rathore, 2023-04-10:
Hi @abmainkar 

Changes for the issue are merged.

Thanks,
ACS team

closed 2023-04-10 by chetan-rathore