ARM-software/bsa-acs #132: PE tests: Incorrectly counting disabled processor entries as available PEs

samerhaj, 2023-03-22:
The BSA PE tests in ACPI mode use the APIC (MADT) table to determine the number of PEs in the system. The code relies on counting the GICC entries in that table: https://github.com/ARM-software/bsa-acs/blob/main/platform/pal_uefi_acpi/src/pal_pe.c#L192

However, there are implementations that list additional entries but mark them as "Disabled" in the GICC Flags Bit0. 

The ACPI 6.5 spec defines that flag as follows:

> "If this bit is set, the processor is ready for use. If this bit is clear and the Online Capable bit is set, the system supports enabling this processor during OS runtime. If this bit is clear and the Online Capable bit is also clear, this processor is unusable, and the operating system support will not attempt to use it."

So BSA ACS should not count disabled entries as valid PEs. In one example, BSA detected "16 PEs" even though the SoC had only 8 PEs. The ACPI table correctly listed 8 GICC entries as enabled, and had an extra 8 entries marked as disabled. This is resulting in failures in multiple PE tests (1, 3, 4, 5, ...) that look like output below. On those same systems, OSes are correctly identifying only 8 online/valid PEs (not 16)

   ```
3 : Check for AdvSIMD and FP support               START
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: success  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       PSCI_CPU_ON: failure[-2]  
       Failed on PE -    8
       B_PE_03
```

chetan-rathore, 2023-03-28:
Hi @samerhaj,

Thanks for pointing out the "Disabled" bit of GICC flags. Code is updated to check for the necessary bits and accordingly add PE in the info table.

Changes are under verification.

Thanks,
ACS team

samerhaj, 2023-03-29:
Verified that this fixes the issue. Thanks @chetan-rathore !

PR #139 (2023-04-03): Only add usable PE in pe_info_table
Fix #132 
 * Check MADT (GICC CPU interface flags) enabled and online capable bit
 * if either of those bits are set, then only add pe
 * if both bits are clear, pe is unusable

Pending: Similiar changes for pal_uefi_dt
closed 2023-04-03 by chetan-rathore