ARM-software/bsa-acs #133: Test 405: interrupt out of order/missed with -v 1 option

foxmmmm, 2023-03-23:
Run v23.01_2.0.0_BETA-0/Bsa.efi with -v 1 option, test 405 will fail and 530 hang:

FS2:\EFI\BOOT\bsa\> Bsa.efi -v 1 -sbsa
 BSA Architecture Compliance Suite
          Version 1.0.3

 Starting tests with print level :  1

..........................

 206 : Check PPI Assignments for OS                   START

       Received interrupt
       Received interrupt
       B_PPI_01
                                       : Result:  PASS
         END

Hypervisor View:
 226 : Check PPI Assignments Hypervisor               START

       Received interrupt
       Received interrupt
       gic version from info table = 4
       gic version from info table = 4
       gic version from info table = 4
       gic version from info table = 4
       Received GIC maintenance interrupt
       B_PPI_02
                                       : Result:  PASS
         END

..........................
 404 : Generate Mem Mapped SYS Timer Intr             START

       Received interrupt
       B_TIME_08
                                       : Result:  PASS
         END

 405 : Restore PE timer on PE wake up                 START

       System timer interrupt received                                ------------------------- QI. Received interrupt before PSCI, why?  
       PSCI VERSION = 10001
       PSCI MAJOR VERSION = 1
       PSCI FEATURS = 2
       PSCI PWR_STATE_FMT = 1
       Received interrupt                                                     -------------------------  QII. Why 405 print this message?
       Read back PE timer count :-871324
       Failed on PE -    0
       B_TIME_09
       Checkpoint --  2                           : Result:  FAIL
         END

      *** One or more tests have Failed/Skipped.***

      *** Starting Power and Wakeup semantic tests ***
Operating System View:
 503 : Wake from EL0 PHY Timer Int                    START  
                                                                                     ------------------------- QIII. Hang here, cause is 405's PCSI?
Here are 3 questions above.

chetan-rathore, 2023-03-28:
Hi @foxmmmm, 

We had some recent changes related to PSCI cmds commited in the mainline. 
Can you please run the latest binary, attached here (https://github.com/chetan-rathore/bsa-acs/blob/main/prebuilt_images/Bsa_ES.efi) with command Bsa_ES.efi -v 1 -sbsa and share the logs.

Thanks,
ACS team

chetan-rathore, 2023-04-12:
Hi @foxmmmm, 

For the issue#133… please find response to the queries.

System timer interrupt received                                ------------------------- QI. Received interrupt before PSCI, why ?
   PSCI VERSION = 10001
   PSCI MAJOR VERSION = 1
   PSCI FEATURS = 2
   PSCI PWR_STATE_FMT = 1
   Received interrupt                                                     -------------------------  QII. Why 405 print this message?
   Read back PE timer count :-871324
   Failed on PE -    0
   B_TIME_09
   Checkpoint --  2                           : Result:  FAIL

Q1 Response)  It looks like system timer expired even before PE entered low power mode, it possible PE took some time entering the low power mode and timeout value was small for system timer.
I have increased the timeout of the System timer in debug binary.

Q2 Response) PE is wake up by the PE timer (el1 phy) …test 206 has installed isr for el1 phy …this message is from test 206 isr handler.
I have add a isr for el1 phy timer in this test itself, in the debug binary

503 : Wake from EL0 PHY Timer Int START
------------------------- QIII. Hang here, cause is 405's PCSI?    
Yes …it could be caused by some corruption due to 405 test interrupt.

Please share a -v 1 log  for issue 133 with this binary (https://github.com/chetan-rathore/bsa-acs/blob/main2/prebuilt_images/Bsa_133.efi)

chetan-rathore, 2023-04-19:
Hi @foxmmmm,

Thanks for the debug logs, changes for #133 and #134 are verified.
Changes are merged now.

Thanks,
ACS team

chetan-rathore, 2023-04-24:
Hi @foxmmmm,

Closing this issue, please reopen if any support is needed.

Thanks,
ACS team