ARM-software/bsa-acs #134: More than 240 nodes of a IORT table

foxmmmm, 2023-03-27:
One system has 388 nodes of the iort table below: 
[image]

If Bsa is run with default maximum 240 nodes, it will be hung:
 BSA Architecture Compliance Suite
          Version 1.0.3

 Starting tests with print level :  3

 Creating Platform Information Tables
 PE_INFO: Number of PE detected       :  256
 GIC_INFO: Number of GICD             :    1
 GIC_INFO: Number of ITS              :   36
 TIMER_INFO: Number of system timers  :    2
 WATCHDOG_INFO: Number of Watchdogs   :    1
 PCIE_INFO: Number of ECAM regions    :    C
 PCIE_INFO: Number of BDFs found      :    29
 PCIE_INFO: Number of RCiEP           :    0
 PCIE_INFO: Number of RCEC            :    0
 PCIE_INFO: Number of EP              :    6
 PCIE_INFO: Number of RP              :   34
 PCIE_INFO: Number of iEP_EP          :    0
 PCIE_INFO: Number of iEP_RP          :    0
 SMMU_INFO: Number of SMMU CTRL       :    20
 Peripheral: Num of USB controllers   :    1
 Peripheral: Num of SATA controllers  :    0
 Peripheral: Num of UART controllers  :    1

      ***  Starting PE tests ***
Operating System View:
   1 : Check Arch symmetry across PE              : Result:  PASS
   2 : Check for number of PE
**.............................................................. Hang here**

chetan-rathore, 2023-03-28:
Thanks @foxmmmm for raising the issue. 

We are checking at our side. If possible can you please also share the acpiview dump of the system.

Thanks,
ACS team

chetan-rathore, 2023-04-17:
Hi @foxmmmm, 

The IOVIRT table size is increased to accommodate 600 iort nodes from the current  240.
The current system had 388 nodes, so the new size should be fine.

Here is the binary with both issues (#133  and #134) fix:  https://github.com/chetan-rathore/bsa-acs/blob/main2/prebuilt_images/Bsa_133_134.efi

Thanks,
ACS team

chetan-rathore, 2023-04-19:
Hi @foxmmmm,

Changes are merged.

Thanks,
ACS team

chetan-rathore, 2023-04-24:
Hi @foxmmmm,

Closing this issue, please reopen if any support is needed.

Thanks,
ACS team

StevenHuang0117, 2023-10-31:
Hi @chetan-rathore 
We have one system has 1476 nodes of the iort table below:
...
  IORT node offset:1C2B8, type: 5
  IORT node offset:F00, type: 4
  IORT node offset:1C2E0, type: 5
  IORT node offset:F6C, type: 4
  IORT node offset:1C308, type: 5
  IORT node offset:F6C, type: 4
  IORT node offset:1C330, type: 5
  IORT node offset:FD8, type: 4
  IORT node offset:1C358, type: 5
  IORT node offset:FD8, type: 4
  IORT node offset:1C380, type: 5
  IORT node offset:1044, type: 4
  IORT node offset:1C3A8, type: 5
  IORT node offset:1044, type: 4
  Number of IOVIRT blocks = 1476
  ITS Group Num ITS: 1
  ITS ID: 2
  ITS Group Num ITS: 1
  ITS ID: 3
  ITS Group Num ITS: 1
  ITS ID: 5
  ITS Group Num ITS: 1
  ITS ID: 6
  ITS Group Num ITS: 1
...

And Bsa will produce some errors below：
...
 228 : Check GIC Maintenance PPI Assignment           START

       Received DAbort Exception Instance 0x23 not found for memory type 0x1001

       Error in obtaining normal memory for instance 35
       Received DAbort Exception Instance 0x24 not found for memory type 0x1001

       Error in obtaining normal memory for instance 36
        Unexpected exception occured
        FAR reported = 0x9CED1028
        ESR reported = 0x8600000E
     ------------------------------------------------------- 
     Total Tests run  =   26  Tests Passed  =   22  Tests Failed =    3
     ------------------------------------------------------- 
ASSERT [DxeCore] /zte/Zhufeng/cpu2.0/uefi/byo_crb/MdeModulePkg/Core/Dxe/Mem/Pool.c(710): Head->Signature == ((('p') | ('h' << 8)) | ((('d') | ('0' << 8)) << 16)) || Head->Signature == ((('p') | ('h' << 8)) | ((('d') | ('1' << 8)) << 16))
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
...

We modify IOVIRT table size to 1M bytes, Bsa run OK.  
The  UEFI Sbsa, Bsa and Sbsa of linux-acs has same problem too.

rajatgoyal47, 2023-11-02:
Hi @StevenHuang0117,

Thank you for reporting this. We will have a look at the issue.
In the meantime, can you please open a new issue for better tracking of the issue.

Thanks,
ACS Team

StevenHuang0117, 2023-11-05:
@rajatgoyal47 

OK.