ARM-software/bsa-acs #135: Is there scenario document for SystemReady SR?

hrw, 2023-03-28:
docs/ directory has scenario documents for SystemReady ES and IR. There is no such one for SystemReady SR (or LS).

I am trying to create new version of my [xBSA checklist page](https://marcin.juszkiewicz.com.pl/download/tables/bsa-sbsa.html) and to add there info which (S)BSA ACS tests need to pass for each entry.

Scenarios cover several tags used by BSA ACS which are not mentioned in BSA specification (DEN0094).

chetan-rathore, 2023-04-01:
Hi @hrw, 

Its a nice suggestion, generally as both ES and SR are very close in terms of BSA rules needs to run on ES or SR systems, ES scenario document is valid for SR also, but with the current names of the documents it might seems that SR guide is missing. 

We will discuss internally on two approaches.
  * Have only two scenario documents ( one for systems that uses Device tree and one for the system using ACPI) + one .md file that reflects which rules are applicable for which band and at what levels (something closer the table mentioned in your checklist page)
  * Have separate scenario documents for each band.

We will keep you updated on the same.

Thanks,
ACS team

chetan-rathore, 2023-06-12:
Hi @hrw,

ACS design makes test layer agnostic to the PAL layer. Majority of test algorithm is same across systems with device tree, acpi table or even when running in baremetal enviroment.

Based on ACS design, Single Test Scenario document will be sufficient. 
Further a testcase checklist will be added, which will cover
* Which test is required for which platforms (IR, ES, SR, baremetal)
* Which tests are verified at ACS end and which are not due to the required hardware not being available?

We are planning to upstream the changes by this month end.

Thanks,
ACS team

chetan-rathore, 2023-07-14:
Hi @hrw, 

As part of BSA ACS 1.0.5 release, we have made slight changes to documentation as discussed. (https://github.com/ARM-software/bsa-acs/commit/b30c93dbf6c239d9df1505d8364a0bffbe58a2f6)

* Single test scenario document covering test algorithm for a test
* testcase checklist which indicates for which Systemready band a test is required to run.

Thanks,
ACS team

chetan-rathore, 2023-07-17:
Thanks @hrw for raising this, we are closing this as changes are merged.

closed 2023-07-17 by chetan-rathore