ARM-software/bsa-acs #138: 205 : Non-secure SGIs are implemented - failure with the non-zero GICR Base Address in MADT GICC structures

sunnywang-arm, 2023-03-31:
205 : Non-secure SGIs are implemented                START
       PE MPIDR 0x0       gicr_rdindex 0       gicrd_base 0x12200000
       PE RD base address 0
       Failed on PE -    0
       B_GIC_05
       Checkpoint --  2                           : Result:  FAIL 
         END

chetan-rathore, 2023-04-01:
Thanks @sunnywang-arm for raising the issue. 

This is observed on system which has non-zero RD base address as part of MADT GICC structure and also has a valid MADT GICR structure. Ideally, when GICR strcuture is present GICC rd base address should be 0. 

ACS will enhance code to ignore GICC rd base address when valid GICR structure is present.

Thanks,
ACS team

chetan-rathore, 2023-04-10:
Hi @sunnywang-arm ,

Changes for the issue are merged.

Thanks,
ACS team