ARM-software/bsa-acs #150: Amount of cards in PCI info block does not match

hrw, 2023-04-27:
My "sbsa-ref" qemu machine has several PCIe/PCI cards plugged in. ACS ignores host bridge ones and legacy PCI ones.

But it does count PCIe/PCI bridges and then amount of listed cards in PCIE INFO block does not match:

```
 PCIE_INFO: Number of ECAM regions    :    1 
  Dev bdf 0x000300 type 0x10 RP bdf 0x000300
  Dev bdf 0x000400 type 0x10 RP bdf 0x000400
  Dev bdf 0x000500 type 0x10 RP bdf 0x000500
  Dev bdf 0x000600 type 0x10 RP bdf 0x000600
  Dev bdf 0x010000 type 0x01 RP bdf 0x000300
  Dev bdf 0x020000 type 0x01 RP bdf 0x000400
  Dev bdf 0x030000 type 0x01 RP bdf 0x000500
  Dev bdf 0x040000 type 0x80 RP bdf 0x000600
 PCIE_INFO: Number of BDFs found      :    8
 PCIE_INFO: Number of RCiEP           :    0 
 PCIE_INFO: Number of RCEC            :    0 
 PCIE_INFO: Number of EP              :    3 
 PCIE_INFO: Number of RP              :    4 
 PCIE_INFO: Number of iEP_EP          :    0 
 PCIE_INFO: Number of iEP_RP          :    0 

  ECAM 0:  Base 0xF0000000
  BDF: 0x300
  Seg: 0x0, Bus: 0x00, Dev: 0x03, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x400
  Seg: 0x0, Bus: 0x00, Dev: 0x04, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x500
  Seg: 0x0, Bus: 0x00, Dev: 0x05, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x600
  Seg: 0x0, Bus: 0x00, Dev: 0x06, Func: 0x0, Dev ID: 0x000C, Vendor ID: 0x1B36
  BDF: 0x10000
  Seg: 0x0, Bus: 0x01, Dev: 0x00, Func: 0x0, Dev ID: 0x000D, Vendor ID: 0x1B36
  BDF: 0x20000
  Seg: 0x0, Bus: 0x02, Dev: 0x00, Func: 0x0, Dev ID: 0x1111, Vendor ID: 0x1234
  BDF: 0x30000
  Seg: 0x0, Bus: 0x03, Dev: 0x00, Func: 0x0, Dev ID: 0x10D3, Vendor ID: 0x8086
  BDF: 0x40000
  Seg: 0x0, Bus: 0x04, Dev: 0x00, Func: 0x0, Dev ID: 0x000E, Vendor ID: 0x1B36

```

8 BDFs found, 3 end point, 4 root port type. One entry is missing - the PCIe/PCI bridge.

My qemu PCI(e) setup:

```
-device pcie-root-port,id=root_port_1.2,chassis=1,slot=2  \
  -device qemu-xhci,id=usb,bus=root_port_1.2 \
-device pcie-root-port,id=root_port_1.3,chassis=1,slot=3  \
  -device bochs-display,bus=root_port_1.3 \
-device pcie-root-port,id=root_port_1.4,chassis=1,slot=4  \
  -device e1000e,bus=root_port_1.4 \
-device pcie-root-port,id=root_port_1.5,chassis=1,slot=5  \
  -device pcie-pci-bridge,id=pci,bus=root_port_1.5 \
    -device virtio-rng-pci,bus=pci,addr=8 \
```

gowthamsiddarthd, 2023-05-05:
Hi @hrw,

Thank you for reporting this. This has been fixed by the PR: https://github.com/ARM-software/bsa-acs/issues/150

Regards,
ACS Team

hrw, 2023-05-05:
```
 PCIE_INFO: Number of ECAM regions    :    1
 PCIE_INFO: Number of BDFs found      :    8
 PCIE_INFO: Number of RCiEP           :    0
 PCIE_INFO: Number of RCEC            :    0
 PCIE_INFO: Number of EP              :    3
 PCIE_INFO: Number of RP              :    4
 PCIE_INFO: Number of iEP_EP          :    0
 PCIE_INFO: Number of iEP_RP          :    0
 PCIE_INFO: Number of UP of switch    :    0
 PCIE_INFO: Number of DP of switch    :    0
 PCIE_INFO: Number of PCI/PCIe Bridge :    0
 PCIE_INFO: Number of PCIe/PCI Bridge :    1
```

closed 2023-05-05 by hrw

hrw, 2023-05-05:
Also for some more complicated setup:

```
Shell> pci
   Seg  Bus  Dev  Func
   ---  ---  ---  ----
    00   00   00    00 ==> Bridge Device - Host/PCI bridge
             Vendor 1B36 Device 0008 Prog Interface 0
    00   00   01    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   00   02    00 ==> Display Controller - VGA/8514 controller
             Vendor 1234 Device 1111 Prog Interface 0
    00   00   03    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   04    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   05    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   06    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   07    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   00   08    00 ==> Bridge Device - Host/PCI bridge
             Vendor 1B36 Device 000B Prog Interface 0
    00   01   00    00 ==> Serial Bus Controllers - USB
             Vendor 1B36 Device 000D Prog Interface 30
    00   02   00    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000C Prog Interface 0
    00   03   00    00 ==> Mass Storage Controller - Serial ATA controller
             Vendor 8086 Device 2922 Prog Interface 1
    00   04   00    00 ==> Network Controller - Ethernet controller
             Vendor 8086 Device 10D3 Prog Interface 0
    00   05   00    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 1B36 Device 000E Prog Interface 0
    00   06   08    00 ==> Pre 2.0 device - UNDEFINED
             Vendor 1AF4 Device 1005 Prog Interface 0
    00   07   00    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 104C Device 8232 Prog Interface 0
    00   08   00    00 ==> Bridge Device - PCI/PCI bridge
             Vendor 104C Device 8233 Prog Interface 0
    00   09   00    00 ==> Multimedia Device - Audio device
             Vendor 8086 Device 2415 Prog Interface 0
Shell> fs0:bsa.efi

 BSA Architecture Compliance Suite
          Version 1.0.4

 Starting tests with print level :  3

 Creating Platform Information Tables
 PE_INFO: Number of PE detected       :    4
 GIC INFO: GIC version                :    v3
 GIC_INFO: Number of GICD             :    1
 GIC_INFO: Number of GICR RD          :    1
 GIC_INFO: Number of ITS              :    1
 TIMER_INFO: Number of system timers  :    0
 WATCHDOG_INFO: Number of Watchdogs   :    1
 PCIE_INFO: Number of ECAM regions    :    1
 PCIE_INFO: Number of BDFs found      :    11
 PCIE_INFO: Number of RCiEP           :    0
 PCIE_INFO: Number of RCEC            :    0
 PCIE_INFO: Number of EP              :    2
 PCIE_INFO: Number of RP              :    6
 PCIE_INFO: Number of iEP_EP          :    0
 PCIE_INFO: Number of iEP_RP          :    0
 PCIE_INFO: Number of UP of switch    :    1
 PCIE_INFO: Number of DP of switch    :    1
 PCIE_INFO: Number of PCI/PCIe Bridge :    0
 PCIE_INFO: Number of PCIe/PCI Bridge :    1
```
[... 37 lines cut ...]
QEMU command (pcie part):

```
-device igb \
-device pcie-root-port,id=root_port_1.2,chassis=1,slot=2  \
  -device qemu-xhci,id=usb,bus=root_port_1.2 \
-device pcie-root-port,id=root_port_1.3,chassis=1,slot=3  \
  -device bochs-display,bus=root_port_1.3 \
-device pcie-root-port,id=root_port_1.4,chassis=1,slot=4  \
  -device e1000e,bus=root_port_1.4 \
-device pcie-root-port,id=root_port_1.5,chassis=1,slot=5  \
  -device pcie-pci-bridge,id=pci,bus=root_port_1.5 \
    -device virtio-rng-pci,bus=pci,addr=8 \
\
\
-device pcie-root-port,id=root_port_1.12,chassis=1,slot=12  \
  -device x3130-upstream,id=upstream_port1,bus=root_port_1.12 \
    -device xio3130-downstream,id=downstream_port1,bus=upstream_port1,chassis=1,slot=20 \
      -device ac97,bus=downstream_port1 \
\
-device pxb-pcie,id=pcie1,bus_nr=2 \
-device pcie-root-port,id=root_port_2.1,bus=pcie1,chassis=2,slot=1 \
  -device ahci,bus=root_port_2.1 \

```