ARM-software/bsa-acs #164: build failed with tag v23.03_REL1.0.4

f4bjh, 2023-05-31:
I'm trying to rebuild tag 23.03_REL1.0.4 of bsa-acs

I get the follwing error : 

gcc  -c  -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../Include/X64/ -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  -Wno-error=stringop-overflow DevicePath.c -o DevicePath.o
cc1: error: -Werror=stringop-overflow: no option -Wstringop-overflow
cc1: error: unrecognized command line option "-Wno-restrict" [-Werror]
cc1: error: unrecognized command line option "-Wno-stringop-truncation" [-Werror]
cc1: all warnings being treated as errors
make[1]: *** [DevicePath.o] Error 1

I'm using tag edk2-stable202208 of edk2 tianocore
I'm using gcc-linaro-7.5.0-2019.12-x86_64_aarch64-linux-gnu.tar.xz

Do you confirm this issue ?

amrathesh, 2023-06-01:
Hi @f4bjh, 

While compiling edk2 Basetools, the build system uses host gcc, In your case it looks like host gcc is unable to cater the flags mentioned. Please try updating host gcc and try again. 

FYI, I got successful compilation with host gcc 11. 
`$ gcc --version
gcc (Ubuntu 11.3.0-1ubuntu1~22.04.1) 11.3.0`

Thanks,
Amrathesh

chetan-rathore, 2023-06-06:
Hi @f4bjh,

Is the build issue resolved after upgrading the  host gcc toolchain at your end?

Thanks,
ACS team