ARM-software/bsa-acs #174: error: conflicting types for ‘val_get_exerciser_err_info’ due to enum/integer mismatch

hrw, 2023-06-20:
Some time passed since I built bsa-acs last time.

```
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/src/acs_exerciser.c:139:10: error: conflicting types for ‘val_get_exerciser_err_info’ due to enum/integer mismatch; have ‘uint32_t(EXERCISER_ERROR_CODE)’ {aka ‘unsigned int(EXERCISER_ERROR_CODE)’} [-Werror=enum-int-mismatch]
  139 | uint32_t val_get_exerciser_err_info(EXERCISER_ERROR_CODE type)
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/src/acs_exerciser.c:19:
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/include/bsa_acs_exerciser.h:121:10: note: previous declaration of ‘val_get_exerciser_err_info’ with type ‘uint32_t(uint32_t)’ {aka ‘unsigned int(unsigned int)’}
  121 | uint32_t val_get_exerciser_err_info(uint32_t type);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

GCC 13.1.1 (default in Fedora 38):

```
16:03 marcin@applejack:code$ aarch64-linux-gnu-gcc --version
aarch64-linux-gnu-gcc (GCC) 13.1.1 20230519 (Red Hat Cross 13.1.1-2)
```

Same problem is in sbsa-acs.

hrw, 2023-06-20:
And there are more of them:

```
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/src/acs_gic.c:406:1: error: conflicting types for ‘val_gic_get_info’ due to enum/integer mismatch; have ‘uint32_t(uint32_t)’ {aka ‘unsigned int(unsigned int)’} [-Werror=enum-int-mismatch]
  406 | val_gic_get_info(uint32_t type)
      | ^~~~~~~~~~~~~~~~
In file included from /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/include/bsa_acs_val.h:21,
                 from /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/src/acs_gic.c:18:
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/include/val_interface.h:106:1: note: previous declaration of ‘val_gic_get_info’ with type ‘uint32_t(GIC_INFO_e)’ {aka ‘unsigned int(GIC_INFO_e)’}
  106 | val_gic_get_info(GIC_INFO_e type);
      | ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

hrw, 2023-06-20:
```
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/src/acs_wd.c:87:1: error: conflicting types for ‘val_wd_get_info’ due to enum/integer mismatch; have ‘uint64_t(uint32_t,  WD_INFO_TYPE_e)’ {aka ‘long long unsigned int(unsigned int,  WD_INFO_TYPE_e)’} [-Werror=enum-int-mismatch]
   87 | val_wd_get_info(uint32_t index, WD_INFO_TYPE_e info_type)
      | ^~~~~~~~~~~~~~~
In file included from /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/include/bsa_acs_val.h:21,
                 from /home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/src/acs_wd.c:18:
/home/marcin/devel/linaro/sbsa-qemu/code/edk2/ShellPkg/Application/bsa-acs/val/include/val_interface.h:190:10: note: previous declaration of ‘val_wd_get_info’ with type ‘uint64_t(uint32_t,  uint32_t)’ {aka ‘long long unsigned int(unsigned int,  unsigned int)’}
  190 | uint64_t val_wd_get_info(uint32_t index, uint32_t info_type);
      |          ^~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```

chetan-rathore, 2023-07-14:
Hi @hrw,

We are working on this, this is observed with gcc 13.1.1 only.

Thanks,
ACS team

PR #186 (2023-07-28): Fix for mismatch in function prototypes
 - Mismatch between function prototype and definition causing build failure with GCC 13.1.1
 - Fixed the function prototype for val_get_exerciser_err_info, val_wd_get_info and val_gic_get_info
 - Resolves: #174
commit 103fa1b3 (2023-07-28): Fix for mismatch in function prototypes

 - Mismatch between function prototype and definition
   causing build failure with GCC 13.1.1
 - Fixed the function prototype for val_get_exerciser_err_info,
   val_wd_get_info and val_gic_get_info
 - Resolves: #174
closed 2023-07-28 by gowthamsiddarthd