ARM-software/bsa-acs #182: Align output of test 304

hrw, 2023-07-17:
```
      *** Starting SMMU tests ***

Operating System View:
 301 : All SMMUs have same Arch Revision          : Result:  PASS
 302 : Check SMMU Granule Support                 : Result:  PASS
 303 : Check SMMU Large Physical Addr Support
       Checkpoint --  1                           : Result:  SKIPPED
 304 : S-EL2 not implemented & SMMU stage1 support     : Result:  PASS

Hypervisor View:
```

rajatgoyal47, 2023-07-26:
Hi Marcin,

We have updated the test description, Thanks for your inputs.

Thanks.

closed 2023-08-01 by rajatgoyal47

hrw, 2023-08-17:
Please configure your editors to remove spare whitespaces...

```diff
- 304 : S-EL2 not implemented & SMMU stage1 support     : Result:  PASS
+ 304 : Check SMMU S-EL2 & stage1 support          : Result:  PASS ^M

 Hypervisor View:
- 352 : S-EL2 not implemented & SMMU stage2 support
+ 352 : Check SMMU S-EL2 & stage2 support     ^M
        SMMUv3.1 not providing Stage2 functionality
```