ARM-software/bsa-acs #183: Test 822 should output expected and read values

hrw, 2023-07-17:
```
 822 : Check Type 1 config header rules
       BDF 0x300 : SLT attribute mismatch
       BDF 0x400 : SLT attribute mismatch
       BDF 0x500 : SLT attribute mismatch
       BDF 0x600 : SLT attribute mismatch
       BDF 0x700 : SLT attribute mismatch
       BDF 0x800 : SLT attribute mismatch
       BDF 0x10000 : SLT attribute mismatch
       Failed on PE -    0
       Checkpoint --  7                           : Result:  FAIL
```

There should be read and expected values of SLT attribute shown. Kind like SBSA ACS does:

```
 825 : Check Device Control register rule
       BDF 0x100 : WARNING PFE attribute mismatch: 0x200 instead of 0x0
       BDF 0x100 : WARNING APPE attribute mismatch: 0x400 instead of 0x0
       BDF 0x200 : WARNING PFE attribute mismatch: 0x200 instead of 0x0
       BDF 0x200 : WARNING APPE attribute mismatch: 0x400 instead of 0x0: Result:  PASS

```

chetan-rathore, 2023-08-08:
Hi @hrw,

The fix is merged for the request.

Thanks,
ACS team

closed 2023-08-08 by chetan-rathore

hrw, 2023-08-17:
```
 822 : Check Type 1 config header rules               START

       Number of bit-field entries to check 10
       BDF 0x300 : SLT attribute mismatch: 0xFF030100 instead of 0x30100
```

thanks