ARM-software/bsa-acs #192: Test 835 - FLR test should not look for BARs to be all 00s

samerhaj, 2023-08-15:
https://github.com/ARM-software/bsa-acs/blob/fafe6eae7750440588b3e70e60ae0daad625d342/test_pool/pcie/operating_system/test_os_p035.c#L30

The BSA Test 835 - PCIe Function Level Reset (FLR) is doing the following:

1. Check for FLR cap if present, proceedd with the test
2. Hit FLR and make sure FLR completes in 100ms 
3. Check the VID is not FF (sanity check)
4. check all BARs are 00s
5. Ensure BME is clear 
6. Ensure MSE clear 

Step 4 is not correct. The code is testing that all of the BAR registers are 0. According to the spec, they could be any value.

Also, Step 1 should print a message indicating thae FLR is supported on device with BDF ...

samerhaj, 2023-08-17:
Issue is confirmed fixed (And test passes) with a BSA debug build (https://github.com/Sujana-M/bsa-acs/commits/main/prebuilt_images/Bsa.efi).

gowthamsiddarthd, 2023-08-21:
HI @samerhaj,

Thank you for confirming the fix. The changes have been raised in PR: https://github.com/ARM-software/bsa-acs/pull/195

Regards,
Gowtham,
ACS Team

closed 2023-08-22 by gowthamsiddarthd