ARM-software/bsa-acs #194: Test 835 - FLR test should wait longer than 100ms and poll for device readiness after reset

samerhaj, 2023-08-18:
[BSA ACS Test 835 ](https://github.com/ARM-software/bsa-acs/blob/main/test_pool/pcie/operating_system/test_os_p035.c#L106) tests PCIe FLR as follows:

- Check if FLR PCIe capability is set
- If so, initiate FLR
- Wait 100 ms
- Read back the VID register from config space
- If value is 0xFF, fail with the message "BDF 0x%x not present"

This results in a failure such as this:

```
835 : Check Function level reset
    BDF 0x2010000 not present
```

The PCIe spec requires that a function completes the FLR operation within 100ms, but subsequent function initialization may need more time. The function is allowed to return Request Retry Status (RRS) Completion Status indicating it needs more time.

The Linux FLR code [pcie_flr() ](https://github.com/torvalds/linux/blob/master/drivers/pci/pci.c#L4753) does the following: 

- Check if FLR PCIe capability is set
- If so, initiate FLR
- Wait 100 ms
- Then call [pci_dev_wait() ](https://github.com/torvalds/linux/blob/master/drivers/pci/pci.c#L1163 )
- pci_dev_wait() polls every 1 ms, until the timeout (which in this case, [PCIE_RESET_READY_POLL_MS](https://github.com/torvalds/linux/blob/master/drivers/pci/pci.c#L80), 60 seconds!)
- For every poll, it reads the PCI_COMMAND register, and looks for FFs ([PCI_POSSIBLE_ERROR](https://github.com/torvalds/linux/blob/master/include/linux/pci.h#L165) ) to determine it should keep polling.

This test should be modified to follow the PCIe spec. One suggestion is to do the following:

- Check if FLR PCIe capability is set
- If so, initiate FLR
- Wait 100 ms
- Read back the VID register from config space
- If value is 0xFF, delay for 1 ms, then read again
- Keep polling until a maximum arbitrary timeout occurs

The Linux code uses 60 seconds because of issues seen in some NVMe drives. For the BSA ACS test, a timeout value of 5 seconds should be is more than enough.

The test can be further enhanced so that instead of assuming FFs is a fialure, check for a completion timeout/UR/CA on seeing all FF’s. If we do not get a completion timeout or other error like UR or CA, then we can assume that the  device returned RRS.

samerhaj, 2023-08-18:
Test confirmed to pass with the fix implemented in test build: https://github.com/Sujana-M/bsa-acs/blob/main/test_image/Bsa_timeout.efi. Thank you @gowthamsiddarthd. Will the patch wit hthe fix get committed ?

gowthamsiddarthd, 2023-08-22:
Hi @samerhaj,

Thank you for confirming the patch. The fix has been raised by the PR: https://github.com/ARM-software/bsa-acs/pull/196

Regards,
Gowtham,
ACS Team

closed 2023-08-23 by gowthamsiddarthd