ARM-software/bsa-acs #199: PE tests 9, 11, 12, and 54 could be more verbose on failure

paulzzy, 2023-08-30:
For each test below, I've linked to the relevant code sections and described what information could be helpful to print out on failure.

PE test 9 (rule B_PE_09):

For failure checkpoint 2 (line 44), it could be helpful to print the number of PMU counters, e.g. by printing out `data`.

https://github.com/ARM-software/bsa-acs/blob/15a223a266f6117747042f50b8ff6bc4f5404aa1/test_pool/pe/operating_system/test_os_c009.c#L35-L45

PE test 11 (rule B_PE_11):

Could be helpful to print out the `breakpointcount` variable when the failure condition is met.

https://github.com/ARM-software/bsa-acs/blob/15a223a266f6117747042f50b8ff6bc4f5404aa1/test_pool/pe/operating_system/test_os_c011.c#L38-L41

PE test 12 (rule B_PE_12):

Could be helpful to print out `(data >> 20) & 0xF` when the failure condition is met, which should represent the number of watchpoints.

https://github.com/ARM-software/bsa-acs/blob/15a223a266f6117747042f50b8ff6bc4f5404aa1/test_pool/pe/operating_system/test_os_c012.c#L34-L37

PE test 54 (B_PE_21):

Could be helpful to print `(data & 0x0F800) >> 11` when the failure condition is met, which should represent the number of PMU counters.

https://github.com/ARM-software/bsa-acs/blob/15a223a266f6117747042f50b8ff6bc4f5404aa1/test_pool/pe/hypervisor/test_hyp_c004.c#L33-L36

chetan-rathore, 2023-09-13:
Hello @paulzzy, 

The PE tests mentioned are multi-PE test, ACS boots the secondary PE and performs the check.
val_print internally calls UEFI AsciiPrint API.  We have observed hang issues when val_print is called from secondary PE.

Generally, all PEs should be architecturally symmetric for BSA/SBSA complaince, so to debug any multi-PE test failure as of now we will enable prints for primary PE.

Thanks,
ACS team

chetan-rathore, 2023-09-19:
Hi @paulzzy, 

As mentioned earlier, tests are updated to print primary PE reg values in case of failure.
We will update sbsa-acs also with similar changes.

Thanks,
ACS team

closed 2023-09-20 by chetan-rathore