ARM-software/bsa-acs #224: Revisit BSA 804 Logic as the address being accessed might be not safe

sunnywang-arm, 2023-10-30:
As offline discussion within the team, the arbitrary read/writes to device NP memory in BSA 804 is not necessarily a legal thing to do. 

It is possible to do these accesses, but you would need to know what addresses would be valid to access without side effects. Unfortunately, this really depends on the device you are accessing, so the safe address may change based on the device, and one of those addresses being accessed could be a magic “start on fire” register.  

Before we find a SAFE address or some other way to check PCI_IN_13 rule, we might need to disable/remove BSA 804.

chetan-rathore, 2023-11-16:
Thanks @sunnywang-arm for raising the ticket.

Tests 804 and 805 performs P/NP BAR memory access checks. The tests perform
* BAR memory  in-range address access
* BAR memory out of range address access

As you also highlighted access to a not a KNOWN address might lead to side-effects on systems. After further internal discussion, we have decided to split the test cases into two parts.
*  In-range address access will be covered for baremetal run of tests
* Out of range address access will be covered for both baremetal and certification run of tests.

We are planning to upstream the changes by next week.

Thanks,
ACS team

chetan-rathore, 2023-12-18:
Hi @sunnywang-arm,

Required changes for the issue are merged in the ACS.

Thanks,
ACS team

chetan-rathore, 2023-12-21:
Hello @sunnywang-arm,

Changes for the issue are merged in the mainline.

Thanks,
ACS team

closed 2023-12-25 by chetan-rathore