ARM-software/bsa-acs #226: More than 1476 nodes of the IORT table

StevenHuang0117, 2023-11-05:
We have one system has 1476 nodes of the iort table below:
...
IORT node offset:1C2B8, type: 5
IORT node offset:F00, type: 4
IORT node offset:1C2E0, type: 5
IORT node offset:F6C, type: 4
IORT node offset:1C308, type: 5
IORT node offset:F6C, type: 4
IORT node offset:1C330, type: 5
IORT node offset:FD8, type: 4
IORT node offset:1C358, type: 5
IORT node offset:FD8, type: 4
IORT node offset:1C380, type: 5
IORT node offset:1044, type: 4
IORT node offset:1C3A8, type: 5
IORT node offset:1044, type: 4
Number of IOVIRT blocks = 1476
ITS Group Num ITS: 1
ITS ID: 2
ITS Group Num ITS: 1
ITS ID: 3
ITS Group Num ITS: 1
ITS ID: 5
ITS Group Num ITS: 1
ITS ID: 6
ITS Group Num ITS: 1
...

And Bsa will produce some errors below：
...
228 : Check GIC Maintenance PPI Assignment START

   Received DAbort Exception Instance 0x23 not found for memory type 0x1001

   Error in obtaining normal memory for instance 35
   Received DAbort Exception Instance 0x24 not found for memory type 0x1001

   Error in obtaining normal memory for instance 36
    Unexpected exception occured
    FAR reported = 0x9CED1028
    ESR reported = 0x8600000E
 ------------------------------------------------------- 
 Total Tests run  =   26  Tests Passed  =   22  Tests Failed =    3
 ------------------------------------------------------- 
ASSERT [DxeCore] /zte/Zhufeng/cpu2.0/uefi/byo_crb/MdeModulePkg/Core/Dxe/Mem/Pool.c(710): Head->Signature == ((('p') | ('h' << 8)) | ((('d') | ('0' << 8)) << 16)) || Head->Signature == ((('p') | ('h' << 8)) | ((('d') | ('1' << 8)) << 16))
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
ASSERT [Bsa] /home/runner/work/arm-systemready/arm-systemready/SR/scripts/edk2/MdePkg/Library/UefiLib/UefiLibPrint.c(202): Buffer != ((void *) 0)
...

We modify IOVIRT table size to 1M bytes, Bsa run OK.
The UEFI Sbsa, Bsa and Sbsa of linux-acs has same problem too.

StevenHuang0117, 2023-11-08:
Hi @chetan-rathore,

Do you have any plan to fix this issue?  I appreciate you fix it as soon as possible.

Thanks,
Steven Huang

rajatgoyal47, 2023-11-08:
Hi @StevenHuang0117,

Thank you for reporting this. 
We are actively working on fixing this issue. we will raise the changes to github soon.

Thanks,
ACS Team

chetan-rathore, 2023-11-22:
Hello @StevenHuang0117,

We have raised the changes to increase the IORT table size to accommodate 2400 nodes.
Can you please use this fork branch: https://github.com/gontumukkala-balaji/bsa-acs  and confirm changes are working on your side?

Thanks,
ACS team

StevenHuang0117, 2023-11-23:
Hello @chetan-rathore ,

Thanks for your job, yes, your changes are working OK in our BSA test. 

Thanks,
Steven Huang

chetan-rathore, 2023-11-23:
Thanks @StevenHuang0117 for confirming the fix.
Changes are merged in the codebase.

closed 2023-11-23 by chetan-rathore