ARM-software/bsa-acs #230: ENABLE_OOB flag

f4bjh, 2023-11-09:
Hi,

I don't know if this is the right place for this, but I feel confused, regarding the ENABLE_OOB flag

When this flag is enabled, TC number 101 seems to be enabled : 
https://github.com/ARM-software/bsa-acs/blob/13fe5b4e86df7b67f5b1184c2916a44fe7a43fca/val/src/acs_memory.c#L130

But my understanding, is that ENABLE_OOB should be enabled only for baremetal application ?

If so, this leads to no enable TC number 101 in UEFI. 
But https://github.com/ARM-software/bsa-acs/blob/v23.07_1.0.5/docs/Arm_BSA_testcase-checklist.rst seemds to enable this TC in UEFI.

Thanks in advance for your help,
Regards,
Fabrice DECROP LONGET

chetan-rathore, 2023-11-16:
Hi Fabrice, 

Your understanding is correct, ENABLE_OOB should be enabled when running BSA tests as baremetal application.
Currently, the BSA ACS supports two methods of running bare-metal application
* On top of UEFI Shell: the ENABLE_OOB flag needs to be enabled for this case.
* On baremetal boot framework:  More details in  https://github.com/ARM-software/bsa-acs/blob/v23.07_1.0.5/docs/Arm_BSA_ACS_Bare-metal_User_Guide.pdf

Can you please let us know your use case?
* In case if you want to run BSA tests on top of UEFI shell with ACPI tables support:  Please follow  https://github.com/ARM-software/bsa-acs/tree/v23.07_1.0.5#to-start-the-acs-build-for-platform-using-acpi-table-perform-the-following-steps

Thanks,
ACS team

f4bjh, 2023-11-17:
I want to run BSA tests on top of UEFI shell with ACPI tables. My current tag is v23.07_1.0.5. I follow https://github.com/ARM-software/bsa-acs/tree/v23.07_1.0.5#to-start-the-acs-build-for-platform-using-acpi-table-perform-the-following-steps

My hw platform is based on fast model from ARM :
```
$ isim_system --version

        SystemC 2.3.3-Accellera --- Jul  9 2022 12:49:02
        Copyright (c) 1996-2018 by all Contributors,
        ALL RIGHTS RESERVED

Fast Models [11.21.18 (Mar 27 2023)]
Copyright 2000-2023 ARM Limited.
All Rights Reserved.

```

My goal is to execute TC 101  Memory Access to Un-Populated addr.

First I got this : 

```
Shell> Bsa.efi -t 101

[...]

      *** Starting Memory Map tests ***  

Operating System View:
 102 : Mem Access Response in finite time    
       USER OVERRIDE VIA SPECIFIC TESTS - Skip Test      
       Checkpoint --  0                           : Result:  SKIPPED 
 103 : PE must access all NS addr space      
       USER OVERRIDE VIA SPECIFIC TESTS - Skip Test      
       Checkpoint --  0                           : Result:  SKIPPED 

[...]
```
Here, TC 101 is not executed

Then, I try to set ENABLE_OOB flag this way : 
```
$ source ShellPkg/Application/bsa-acs/tools/scripts/acsbuild.sh ENABLE_OOB 
```
I got this : 
```
Processing meta-data 
.Architecture(s)  = AARCH64
Build target     = DEBUG
Toolchain        = GCC49

Active Platform          = edk2/ShellPkg/ShellPkg.dsc
Active Module            = edk2/ShellPkg/Application/bsa-acs/baremetal_app/BsaAcs.inf
. done!

build.py...
 : error F002: Module for [AARCH64] is not a component of active platform. Please make sure that the ARCH and inf file path are given in the same as in [edk2/ShellPkg/ShellPkg.dsc]
	edk2/ShellPkg/Application/bsa-acs/baremetal_app/BsaAcs.inf

- Failed -
Build end time: 11:55:38, Nov.17 2023
Build total time: 00:00:01

Build failed: error while running source at line 9 in ShellPkg/Application/bsa-acs/tools/scripts/acsbuild.sh for [][].
```

Thanks in advance for your help.
Regards,
Fabrice DECROP LONGET

chetan-rathore, 2023-11-20:
Hello @f4bjh,

Thanks for sharing the details.

Test 101 is not enabled for UEFI + ACPI environment, hence it didn't run for binary which was complied using   https://github.com/ARM-software/bsa-acs/tree/v23.07_1.0.5#to-start-the-acs-build-for-platform-using-acpi-table-perform-the-following-steps link.

Test 101 can be run on bare-metal environment, using https://github.com/ARM-software/bsa-acs/tree/v23.11_BootFramework tag.
Please use https://github.com/ARM-software/bsa-acs/tree/v23.11_BootFramework#acs-build-steps---bare-metal-abstraction for building and running the bsa binary on BM environment.

Thanks,
ACS team

chetan-rathore, 2023-11-29:
Hello @f4bjh,

Hope we answered your query. Please reopen the ticket if any clarification/help needed.

Thanks,
ACS team

closed 2023-11-29 by chetan-rathore