ARM-software/bsa-acs #231: Can spaces/tabs be consistent?

hrw, 2023-11-13:
As part of my work on SBSA Reference Platform in QEMU I am running (S)BSA ACS ~weekly. And amount of space related changes suggests very low review quality inside Arm.

Can you get spaces/tabs to be consistent? And then keep it? Do some CI runs on internal review and expose differences in output compared to previous version.

[image]

chetan-rathore, 2023-11-16:
Hi @hrw,

Thanks for raising the issue.

We will take the feedback and remove the end of line spaces and also include same in CI.
Please note the test description uses max 70 character and in cases where the test description is not filling all 70 characters..spaces are used.

Can you please also pass the logs where the "ECAM base is zero " is coming.

Thanks,
ACS team

hrw, 2023-11-16:
https://github.com/hrw/sbsa-ref-status/ has all my logs.

Will check tomorrow are they up to date.

hrw, 2023-11-16:
There is also a script I use to gather logs.

hrw, 2023-11-16:
updated logs

chetan-rathore, 2023-11-23:
Hello @hrw,

We've made changes to eliminate trailing spaces at the end of lines, the numbers of files affected are 54 files. Since these modifications are related to a single issue, our intention is to consolidate them into a single pull request.

Thanks,
ACS team

closed 2023-11-24 by chetan-rathore

hrw, 2023-11-27:
https://github.com/hrw/sbsa-ref-status/commit/344b8a35a21a7782237aa59b8880b4d0bbe5cd40 shows how it changed (fresh ACS logs)

hrw, 2023-11-28:
```
 501 : Wake from EL1 PHY Timer Int           
       START       Received EL1 PHY interrupt

       B_WAK_01, B_WAK_02, B_WAK_03, B_WAK_04, B_WAK_05                     HERE
       B_WAK_06, B_WAK_07, B_WAK_10, B_WAK_11
                                       : Result:  PASS
       END
```

"HERE" shows where extra spaces were left.

commit 1884abe0 (2023-11-24): Removing unwanted whitespaces in console prints (#236)

* Removing unwanted whitespaces in console prints

 - Resolves: #231
 - Removed unwanted whitespace characters before new line in console prints.

* Minor improvements to print messages.

 - Aligned whitespaces and \n with rest of the prints for hygiene logs.
 - Adjusted the code to have max 100 chars in a line.

---------