ARM-software/bsa-acs #234: Can tests FAIL if there are no system timers?

hrw, 2023-11-21:
SBSA Reference Platform in QEMU does not have memory mapped system timers. BSA ACS recognize it and inform in a header:

```
 Creating TIMER INFO table
  GTDT is at FBF8E618 and length is 84
  GTDT revision is at 3
 TIMER_INFO: Number of system timers  :    0
 Creating WATCHDOG INFO table
  Watchdog base = 0x50011000 INTID = 0x30
 WATCHDOG_INFO: Number of Watchdogs   :    1
```

But then tests 403, 404, 405 and 505 are skipped:

```
 403 : Memory mapped timer check                      START

       No System timers are defined
       B_TIME_07, B_TIME_10
       Checkpoint --  1                           : Result:  SKIPPED
         END

 404 : Generate Mem Mapped SYS Timer Intr             START

       No System timers are defined
       B_TIME_08
       Checkpoint --  1                           : Result:  SKIPPED
         END

 405 : Restore PE timer on PE wake up                 START

       No non-secure systimer implemented
       B_TIME_09
       Checkpoint --  1                           : Result:  SKIPPED
         END
 
 505 : Wake from System Timer Int                     START

       No system timers implemented
       B_WAK_01, B_WAK_02, B_WAK_03, B_WAK_04, B_WAK_05
       B_WAK_06, B_WAK_07, B_WAK_10, B_WAK_11
       Checkpoint --  1                           : Result:  SKIPPED
         END
```

Shouldn't they FAIL instead?

BSA spec uses "must":

> B_TIME_07: The system wakeup timer must in the form of the memory mapped timer that is described in the Armv8 ARM

> B_TIME_08: On timer expiry, the system wakeup timer must generate an interrupt that must be wired to the GIC as an SPI or LPI. Also, the system wakeup timer can be used to wake up PEs.

On the other side it also says:

> B_TIME_06: Unless all of the local PE timers are always on, the base system must implement a system wakeup timer that can be used when PE timers are powered down.

And we pass test 403 which checks for it. And that's suggests to me that we can ignore B_TIME_07/08 test results.

chetan-rathore, 2023-11-23:
Hello @hrw,

Your understanding is correct.
If the Processing Element (PE) maintains local timers that are consistently active, the system wakeup timer implementation is optional. Hence, the tests (403,404,405,505)  would skip rather than result in failure.

Thanks,
ACS team

hrw, 2023-11-23:
Thanks.

Looks like we are getting close to SBSA level 3 requirements. Will have to check what else we miss and what can be ignored.

closed 2023-11-23 by hrw