ARM-software/bsa-acs #237: Question about SMMU tests (again)

hrw, 2023-11-24:
When Secure-EL2 is implemented, B_SMMU_08, B_SMMU_16, B_SMMU_17, B_SMMU_18 tests (304, 352) are skipped.

For B_SMMU_08 BSA 1.0C says:

> If Secure-EL2 is not implemented, stage 1 System MMU functionality that is made visible to an operating system must present the interface of a System MMU compatible with one of the following:
>
> - The SMMUv2 specification, where each context bank must present a unique physical interrupt to the GIC.
> - The Arm SMMUv3 specification or higher, where the integration of the System MMUs is compliant with the requirements in Section D.

And B_SMMU_16:

> If a device is assigned and passed through to an operating system under a hypervisor, then the memory transactions of the device must be subject to stage 2 translation, allocation of memory attributes, and application of permission checks, under the control of the hypervisor.
> This specification collectively refers to this translation, attribution, and permission checking as policing. The act of policing is called stage 2 System MMU functionality.

B_SMMU_17:

> From a hardware perspective, this means that a base system supporting a protection hypervisor requires all non-secure DMA capable devices that will be assigned to a non-secure VM for direct control to be policed by stage 2 System MMU functionality.

B_SMMU_18:

> If Secure-EL2 is not implemented, stage 2 System MMU functionality must be provided by a System MMU compatible with the Arm SMMUv2 specification or Arm SMMUv3 specification.

I start to wonder which stage functionality is required when Secure-EL2 is implemented.

Would say that stage1 due to _08 and stage2 due to _16/17 but ACS do not check for any of them when SEL2 is enabled.

chetan-rathore, 2023-12-14:
Hello @hrw,

Thanks for the query, it did lead to a good discussion on our side.

For the main intent of your query "When S-EL2 is enabled, how the stage1 and stage 2 SMMU requirements are checked in BSA-ACS"

BSA does not mandate either Stage 1 or Stage 2 but does mandate certain requirements if these are supported.
If Secure-EL2 is implemented then an SMMUv3.2 is supported, which implies that both SMMU Stage 1 and Stage 2 are implemented. In this case, rule B_SMMU_21 applies for BSA.

Also, the BSA test for B_SMMU_08/18, should also include the SMMU version check when PE supports S-EL2.

Thanks,
Chetan

chetan-rathore, 2023-12-18:
Hello @hrw,

Required changes as discussed above are merged in the ACS.

Thanks,
ACS team

hrw, 2023-12-18:
sbsa-ref with neoverse-n2 on recent QEMU:

```
      *** Starting SMMU tests ***

Operating System View:
 301 : All SMMUs have same Arch Revision          : Result:  PASS
 302 : Check SMMU Granule Support                 : Result:  PASS
 303 : Check SMMU Large Physical Addr Support
       Checkpoint --  1                           : Result:  SKIPPED
 304 : Check SMMU S-EL2 & stage1 support          : Result:  PASS

Hypervisor View:
 352 : Check SMMU S-EL2 & stage2 support
       SMMUv3.1 not providing Stage2 functionality
       Failed on PE -    0
       Checkpoint --  2                           : Result:  FAIL
 354 : SMMUv3 Integration compliance              : Result:  PASS

      One or more SMMU tests failed or were skipped.
```

Emulated SMMU v3.1 has either Stage1 or Stage2 but not both at same time. It is one of remaining blockers for SBSA level 3 compliance (looking at (S)BSA ACS test results).

chetan-rathore, 2024-02-19:
Hi @hrw,

Please let us know if any further help required for this issue from ACS team.

Thanks,
ACS team

closed 2024-02-19 by hrw