ARM-software/bsa-acs #253: B_PE_01 Fail -- The values of PMCEID0_EL0, PMCR_EL0 are different on cores with different architectures in DynamIQ SoC

magicse7en, 2024-01-16:
Since the SoC is DynamIQ architecture(e.g., cortex-a720 + cortex-x4),  the values of PMCEID0_E0 and PMCR_EL0 differ between cortex-a720 and cortex-x4, causing the B_PE_01 test to fail.

Reg compare failed for PE index=4 for Register: PMCEID0_EL0
Current PE value = 0xF0F1A7F7BFF7F3F         Other PE value = 0xF0F1A7F7FFF6F3F
Failed on PE -    4
B_PE_01
Checkpoint --  1                           : Result:  FAIL 
  END

Any ideas?

samerhaj, 2024-01-16:
The first rule in the BSA specification states that:

"RB_PE_01 All PEs are architecturally symmetric except for the permitted exceptions listed in Section A."

Section A has table 18, which shows the permitted differences for systems composed of heterogeneous PEs, such as big.LITTLE. These variations are not expected to be perceived as architectural differences by a general purpose OS.

PMCEID0_E0 and PMCR_EL0 are not in that list, which explains the failure.

magicse7en, 2024-01-16:
> The first rule in the BSA specification states that:
>
> "RB_PE_01 All PEs are architecturally symmetric except for the permitted exceptions listed in Section A."
>
> Section A has table 18, which shows [...]

Thanks for the reply.
Yes, I noticed the table 18 in the BSA specification. However, after reading the cortex-a720 and cortex-x4 TRMs:
- cortex-x4 supports ID26 (refer to Table A-332 in section A7.3 of TRM), but cortex-a720 doesn't support it (refer to Table A-251 in section A5.3)
- cortex-x4 doesn't support ID12, but cortex-a720 supports it.
I'm just curious if this heterogeneous PEs like cortex-a720+cortex-x4 are not BSA compliant?

sunnywang-arm, 2024-06-07:
@magicse7en 
Could you also provide the value of PMCR_EL0 in cortex-a720 and cortex-x4? there is no details about PMCR_EL0 so we don't know where the difference is.

magicse7en, 2024-06-17:
> @magicse7en Could you also provide the value of PMCR_EL0 in cortex-a720 and cortex-x4? there is no details about PMCR_EL0 so we don't know where the difference is.

The main difference is that the PMCR_EL0 bit[15:11] (indicate the number of event counters implemented), the number of event counters of cortex-a720 and cortex-x4 are diffferent.

sunnywang-arm, 2024-06-17:
Thanks, @magicse7en 

According to the TRMs for a720 and x5 (the links below), PMCR_EL0 bit[15:11] are reserved and the reset values for both a720 and x5 PMCR_EL0 bit[15:10] are the same (all 0s). Could you confirm that the difference is in [15:11]? It looks like the difference could only occur in bits [9][7][5].  
- https://developer.arm.com/documentation/102530/0001/External-registers/External-PMU-registers-summary/PMCR-EL0--Performance-Monitors-Control-Register
- https://developer.arm.com/documentation/102484/0001/ext-registers/External-PMU-registers-summary/PMCR-EL0--Performance-Monitors-Control-Register

magicse7en, 2024-06-19:
Hi @sunnywang-arm,
I think the PMCR_EL0 is the AArch64_pmcr_el0, not ext_pmcr_el0. right?
- https://developer.arm.com/documentation/102484/0001/AArch64-registers/AArch64-Performance-Monitors-registers-summary/PMCR-EL0--Performance-Monitors-Control-Register?lang=en
- https://developer.arm.com/documentation/102530/0001/AArch64-registers/AArch64-Performance-Monitors-registers-summary/PMCR-EL0--Performance-Monitors-Control-Register?lang=en

The A720 PMCR_EL0 value: 0xA0E0
The X4 PMCE_EL0 value: 0xF8E0

sunnywang-arm, 2024-06-20:
Yes, you're right. Thanks for pointing that out, @magicse7en. We'll also check this.

hrw, 2024-11-20:
On RK3588 (A76 + A55) registers also differ on PE 4-7:

```
        PE Index: 4, ID_AA64PFR0_EL1    : 0x1100000011111112    FAIL
          Masked Primary PE Value : 0x0000000011112222 
          Masked Current PE Value : 0x1100000011111112 

        PE Index: 4, PMCEID1_EL0        : 0x0000000000F2AE7F    FAIL
          Masked Primary PE Value : 0x0000000001F0AE7F 
          Masked Current PE Value : 0x0000000000F2AE7F 
        PE Index: 4, PMCR_EL0           : 0x0000000000000000    FAIL
          Masked Primary PE Value : 0x0000000041453000 
          Masked Current PE Value : 0x0000000000000000 

        PE Index: 4, ID_PFR1_EL1        : 0x10010000    FAIL
          Masked Primary PE Value : 0x10011011 
          Masked Current PE Value : 0x10010000 
```

PE 0-3 are A55, 4-7 are A76

samerhaj, 2025-04-22:
This is similar to #433 

This is being discussed for the BSA spec itself. As of today, some OSes can handle these differences across PEs as long as groups of PEs of the same MIDR have identical values. Other OSes do not handle this gracefully.

hrw, 2025-04-22:
You mean #438 rather than #433