ARM-software/bsa-acs #254: BSA 102 installing exception handler at EL2, even if running at EL1

PhatFree, 2024-02-09:
BSA 102 is testing Memory Response in a finite amount of time, the test is expected to potentially cause an exception. So BSA installs its own exception handler (using the RegisterInterruptHandler of the UEFI CPU Arch Protocol), but is doing so at EL2. 
In a VM case, UEFI may be running at EL1/EL0, and the VM firmware (in the hypervisor) has its own exception handler installed, which may triggers a system reboot since the BSA handler doesn't handle the exception.

 BSA should check the EL level its running at, and install its exception handler at same EL level.

chetan-rathore, 2024-02-17:
Hi @PhatFree, 

As discussed internally, ACS installs the exception handler at the EL on which it is running.

Thanks,
ACS team

chetan-rathore, 2024-02-19:
Hi @PhatFree, 

We will open the issue once the ESR value dump is received for further checking why exception was routed to EL2.

Thanks,
ACS team

closed 2024-02-19 by chetan-rathore