ARM-software/bsa-acs #256: BSA detects some PCIe devices but not list them as valid device for testing

sunnywang-arm, 2024-02-15:
The message below showed zero BDF found but it still be able to print out the class code info for BDF 0x0, 0x100, 0x200, and all PCI device tests got skipped. 

```
 PCIE_INFO: Number of ECAM regions    :    1 
 PCIE_INFO: Number of BDFs found      :    0
   BDF Table: No RCiEP or iEP found
.....
 
 BDF is 0
 Class code is 2000001
 BDF is 100
 Class code is FF0000
 BDF is 200
 Class code is 1080201 
 Peripheral: Num of Network ctrl      :    1
 Peripheral: Num of Storage ctrl      :    1 
... 
 .....
      *** No Valid Devices Found,                 Skipping remaining PCIE tests *** 
```

chetan-rathore, 2024-02-17:
Hi @sunnywang-arm,

In the SUT, there were only legacy PCI devices present. Having PCIe devices are prerequisite for running ACS PCIe tests else ACS will skip the PCIe tests with the mentioned log message.

For the peripheral info table, ACS considers both legacy and PCIe devices.

We will add log prints to list the reason of BDF skip during creation of PCIe info table to help in debugging.

Thanks,
ACS team

chetan-rathore, 2024-02-19:
Hi @sunnywang-arm,

Thanks for raising this.  
We have added DEBUG log prints to capture the reason of skipping a PCIe device from Info Table.
This will ensure all BDF processed by ACS are matching with uefi pci command.

Log snippet:
PCIE_INFO: Number of ECAM regions    :    1
       **BDF 0x0 is a Host Bridge...Skipping
       BDF 0x900 is a Host Bridge...Skipping
       BDF 0x20000 PCI Express capability not present...Skipping
       BDF 0x80900 PCI Express capability not present...Skipping**
  Dev bdf 0x000100 type 0x200 RP bdf 0xFFFFFFFF
  Dev bdf 0x000200 type 0x200 RP bdf 0xFFFFFFFF
  Dev bdf 0x000300 type 0x10 RP bdf 0x000300
  Dev bdf 0x000400 type 0x10 RP bdf 0x000400
  Dev bdf 0x000500 type 0x10 RP bdf 0x000500
  Dev bdf 0x000600 type 0x10 RP bdf 0x000600
  Dev bdf 0x000700 type 0x10 RP bdf 0x000700
  Dev bdf 0x000800 type 0x10 RP bdf 0x000800
  Dev bdf 0x010000 type 0x10 RP bdf 0x010000
  Dev bdf 0x030000 type 0x01 RP bdf 0x000400
  Dev bdf 0x040000 type 0x01 RP bdf 0x000500
  Dev bdf 0x050000 type 0x01 RP bdf 0x000600
  Dev bdf 0x060000 type 0x01 RP bdf 0x000700
  Dev bdf 0x070000 type 0x80 RP bdf 0x000800

PR #257 (2024-02-19): Added Debug message to print BDF skip reason
Fixes #256

Added debug prints to log message when BDF is skipped due to
 * PCI Express capability not present
 * BDF is host bridge
 * Platform input as marked BDF as invalid.
closed 2024-02-19 by chetan-rathore